raspberry-vanilla/android_local_manifest #110: Dialer app and Bluetooth media app not working on android-14.0.0_r22  or android-14.0.0_r52 for aosp_rpi4_car-userdebug or lunch aosp_rpi4_car-ap2a-userdebug

zenanswer, 2024-07-19:
I try to build one car (AAOS) image following those steps below (using r22 as a sample).
The AAOS on Pi4 (8G RAM) could startup, and could pair with a Pixel 3(Android 12), or iPhone 15 (iOS 16).

When I try to start up the dialer app or bluetooth media app, "connect to a Bluetooth device" shows up on the screen.
And cannot hear anything through the 3.5mm headset on Pi4, when playing music on those Phones

On Pixel 3, playing some music, and adjust the volume, I can see the bluetooth icon on the volume bar.
On iPhone, I do select the Pi4 as the "speaker".

I noticed #33  and https://github.com/raspberry-vanilla/android_device_brcm_rpi4/pull/2.
And the modification of "aosp_rpi4_car.mk" has been merged into "android-14.0.0_r22".

So, how can I solve this problem?

Hardware: Pi4 8GB RAM, type-C port to PC USB3, HIMD0 to a PC monitor, no extra power supply, Logitech wireless mouse.

Compile steps:
```
repo init -u https://android.googlesource.com/platform/manifest -b android-14.0.0_r22 --depth=1
curl -o .repo/local_manifests/manifest_brcm_rpi.xml -L https://raw.githubusercontent.com/raspberry-vanilla/android_local_manifest/android-14.0.0_r22/manifest_brcm_rpi.xml --create-dirs
curl -o .repo/local_manifests/remove_projects.xml -L https://raw.githubusercontent.com/raspberry-vanilla/android_local_manifest/android-14.0.0_r22/remove_projects.xml

repo sync
. build/envsetup.sh
lunch aosp_rpi4_car-userdebug
make installclean
make bootimage systemimage vendorimage -j4
./rpi4-mkimg.sh
```bash

RahulKalekar, 2024-07-19:
Same issue here

zenanswer, 2024-07-22:
I got some Bluetooth LE error in logcat, so I disabled the GATT in aosp_rpi4_car.mk, and re-tested.
But did not help.

[Bluetooth Low Energy](https://source.android.com/docs/core/connect/bluetooth/ble)

```log
07-19 14:25:10.003  3171  3203 I bluetooth: packages/modules/Bluetooth/system/gd/metrics/metrics_state.cc:161 UploadLEConnectionSession: LEConnectionMetricsRemoteDevice: The session is uploaded for xx:xx:xx:xx:7b:35
07-19 14:25:10.003  3171  3203 I bluetooth: packages/modules/Bluetooth/system/gd/metrics/metrics_state.cc:177 AddStateChangedEvent: LEConnectionMetricsRemoteDevice: Transaction State 0x00000003, Connection Type 0x00000001, Origin Type 0x00000000
07-19 14:25:10.003  3171  3203 I bluetooth: packages/modules/Bluetooth/system/gd/metrics/metrics_state.cc:183 AddStateChangedEvent: LEConnectionMetricsRemoteDevice: Empty Address Cancellation 0x00000003, 0x00000001, 0x00000003
07-19 14:25:10.003  3171  3203 I bluetooth: packages/modules/Bluetooth/system/gd/hci/acl_manager/le_impl.h:1027 disarm_connectability: Disarming LE connection state machine with create connection cancel
07-19 14:25:10.003  3171  3203 W bluetooth: packages/modules/Bluetooth/system/main/shim/acl.cc:1769 OnLeConnectFail: Connection failed le remote:b0:2a:43:fc:7b:35[PUBLIC_DEVICE_ADDRESS]
07-19 14:25:10.003  3171  3197 I bluetooth: packages/modules/Bluetooth/system/stack/arbiter/acl_arbiter.cc:90 OnLeDisconnect: Notifying Rust of LE disconnection
07-19 14:25:10.003  3171  3197 I bt      : bluetooth_core::gatt::arbiter: processing disconnection on transport TransportIndex(0)
07-19 14:25:10.003  3171  3197 I bluetooth: packages/modules/Bluetooth/system/stack/eatt/eatt_impl.h:914 disconnect:  Device: xx:xx:xx:xx:7b:35, cid: 0xffff
07-19 14:25:10.003  3171  3197 W bt_stack: [WARNING:eatt_impl.h(918)] disconnect no eatt device found
07-19 14:25:10.004  3171  3197 I bluetooth: packages/modules/Bluetooth/system/stack/gap/gap_ble.cc:304 client_connect_cback: No active GAP service found for peer:xx:xx:xx:xx:7b:35 callback:Disconnected
07-19 14:25:10.004  3171  3197 I bt_bta_gattc: packages/modules/Bluetooth/system/bta/gatt/bta_gattc_act.cc:1310 bta_gattc_conn_cback: Disconnected att_id:3 addr:xx:xx:xx:xx:7b:35, transport:BT_TRANSPORT_LE reason:UNKNOWN[255]
07-19 14:25:10.004  3171  3197 W bluetooth: packages/modules/Bluetooth/system/stack/acl/ble_acl.cc:153 acl_ble_connection_fail: LE connection fail peer:xx:xx:xx:xx:7b:35[public] bd_addr:xx:xx:xx:xx:7b:35 hci_status:HCI_ERR_HOST_TIMEOUT
07-19 14:25:10.004  3171  3197 W bt_stack: [WARNING:bta_gattc_act.cc(360)] bta_gattc_open_fail: Cannot establish Connection. conn_id=000000. Return GATT_ERROR(133)
07-19 14:25:10.004  3171  3197 I bluetooth: packages/modules/Bluetooth/system/bta/dm/bta_dm_main.cc:96 bta_dm_search_sm_execute: state:BTA_DM_DISCOVER_ACTIVE, event:BTA_DM_DISCOVERY_RESULT_EVT[0x207]
07-19 14:25:10.004  3171  3197 W bt_bta_dm: packages/modules/Bluetooth/system/bta/dm/bta_dm_act.cc:1460 bta_dm_search_cmpl: GATT active but no BLE connection, likely disconnected midway through
```

```makefile
xcwang@xcwang-B760M-POWER:~/dev/aosp/aosp/device/brcm/rpi4$ git diff
diff --git a/aosp_rpi4_car.mk b/aosp_rpi4_car.mk
index 656f3f0..d122bcd 100644
--- a/aosp_rpi4_car.mk
+++ b/aosp_rpi4_car.mk
@@ -33,6 +33,7 @@ PRODUCT_VENDOR_PROPERTIES += \
     bluetooth.profile.opp.enabled=false \
     bluetooth.profile.pbap.server.enabled=false \
     bluetooth.profile.sap.server.enabled=false \
+    bluetooth.profile.gatt.enabled=false \
     bluetooth.profile.vcp.controller.enabled=false

 # Broadcast radio
xcwang@xcwang-B760M-POWER:~/dev/aosp/aosp/device/brcm/rpi4$
```

zenanswer, 2024-12-24:
update: BT audio is working on Android 15 R6 on Rpi4 and Rpi5, but Dialer and Contacts still not working.

aladin153, 2025-02-21:
Hello,

Permissions are not granted automatically by the system for the dialer app.
Try to manually grant all the necessary permissions.

closed 2025-03-18 by KonstaT