raspberry-vanilla/android_local_manifest #116: #error "LONG_BIT definition appears wrong for platform (bad gcc/glibc config?)."

RankoR, 2024-08-31:
Build fails on Arch Linux:

```
/usr/include/python3.12/pyport.h:586:2: error: "LONG_BIT definition appears wrong for platform (bad gcc/glibc config?)."
  586 | #error "LONG_BIT definition appears wrong for platform (bad gcc/glibc config?)."
      |  ^
In file included from src/py/libcamera/py_formats_generated.cpp:12:
In file included from /usr/lib/python3.12/site-packages/pybind11/include/pybind11/pybind11.h:13:
In file included from /usr/lib/python3.12/site-packages/pybind11/include/pybind11/detail/class.h:12:
In file included from /usr/lib/python3.12/site-packages/pybind11/include/pybind11/attr.h:13:
/usr/lib/python3.12/site-packages/pybind11/include/pybind11/detail/common.h:1047:5: error: cannot use 'throw' with exceptions disabled
 1047 |     throw std::runtime_error(reason);
      |     ^
/usr/lib/python3.12/site-packages/pybind11/include/pybind11/detail/common.h:1051:5: error: cannot use 'throw' with exceptions disabled
 1051 |     throw std::runtime_error(reason);
      |     ^
/usr/lib/python3.12/site-packages/pybind11/include/pybind11/detail/common.h:1227:5: error: cannot use 'try' with exceptions disabled
 1227 |     try {
      |     ^
In file included from src/py/libcamera/py_formats_generated.cpp:12:
In file included from /usr/lib/python3.12/site-packages/pybind11/include/pybind11/pybind11.h:13:
In file included from /usr/lib/python3.12/site-packages/pybind11/include/pybind11/detail/class.h:12:
In file included from /usr/lib/python3.12/site-packages/pybind11/include/pybind11/attr.h:14:
In file included from /usr/lib/python3.12/site-packages/pybind11/include/pybind11/cast.h:15:
In file included from /usr/lib/python3.12/site-packages/pybind11/include/pybind11/detail/type_caster_base.h:12:
/usr/lib/python3.12/site-packages/pybind11/include/pybind11/pytypes.h:863:9: error: cannot use 'throw' with exceptions disabled
```

```
$ python --version                                                                                                                                                                                                       
Python 3.12.5
```

I understand the recommended system is Ubuntu, but vanilla AOSP (of any version) builds fine on Arch.

KonstaT, 2024-08-31:
Seems to be an issue with compiling libcamera fork that is part of Raspberry Vanilla project but not part of AOSP.

Sorry, no much clue with the issue itself. Compiles fine on Ubuntu 22.04 LTS which has Python 3.10.12.

RankoR, 2024-09-01:
Looks like some package incompatibility.
I didn't manage to build it from arch directly, but created a docker builder instead:

```
FROM ubuntu:22.04

ENV DEBIAN_FRONTEND=noninteractive

RUN apt-get update && \
    apt-get dist-upgrade -y && \
    apt-get install -y bc coreutils dosfstools e2fsprogs fdisk kpartx mtools ninja-build pkg-config python3-pip rsync && \
    apt-get install -y git-core gnupg flex bison build-essential zip curl zlib1g-dev libc6-dev-i386 x11proto-core-dev libx11-dev lib32z1-dev libgl1-mesa-dev libxml2-utils xsltproc unzip fontconfig && \
    apt-get install -y repo xxd apt-utils openjdk-8-jdk openjdk-11-jdk iputils-ping autoconf openssh-server bsdmainutils simg2img jq && \
    apt-get autoremove -y

RUN pip3 install meson mako jinja2 ply pyyaml dataclasses

RUN useradd -m -u 1000 user

USER 1000:1000

ENTRYPOINT ["/bin/bash"]
```

closed 2024-09-01 by RankoR