raspberry-vanilla/android_local_manifest #118: [    1.993668] init: bool android::init::BlockDevInitializer::InitDevices(std::set<std::string>): partition(s) not found in /sys, waiting for their uevent(s): mmcblk0p2, mmcblk0p3

Marietto2008, 2024-09-01:
Hello bro.

I'm trying to emulate Android 14 for raspberry pi 4 on my PC,where I have installed Ubuntu 24.04 X64 bit using QEMU emulator version 9.0.93 (v9.1.0-rc3). This is the script that I ran :

```
nano start-rpi4 :

qemu-system-aarch64 \
        -smp 4 \
        -m 2048 \
        -cpu cortex-a72 \
        -M virt \
        -device virtio-gpu-pci,xres=1024,yres=768 \
        -display sdl,gl=on \
        -device usb-ehci \
        -device usb-kbd \
        -device virtio-tablet-pci \
        -usb \
        -serial stdio \
        -kernel /mnt/zroot-133/_OS/Android/boot/Image \
        -initrd /mnt/zroot-133/_OS/Android/rpi4/ramdisk.img \
        -drive index=0,if=none,id=system,file=/mnt/zroot-133/_OS/Android/rpi4/system.img \
        -device virtio-blk-pci,drive=system \
        -drive index=1,if=none,id=vendor,file=/mnt/zroot-133/_OS/Android/rpi4/vendor.img \
        -device virtio-blk-pci,drive=vendor \
        -append "console=ttyAMA0,38400 drm.debug=0x0 rootwait rootdelay=5 root=/dev/mmcblk1p2 androidboot.hardware=rpi4 androidboot.selinux=permissive security=selinux selinux=1 androidboot.qemu.hw.mainkeys=0 androidboot.lcd.density=160"
```

This is what happens when I run it :

```
root@Z390-AORUS-PRO-DEST:/mnt/zroot-133/_OS/Android# ./start-rpi4

WARNING: Image format was not specified for '/mnt/zroot-133/_OS/Android/rpi4/system.img' and probing guessed raw.
         Automatically detecting the format is dangerous for raw images, write operations on block 0 will be restricted.
         Specify the 'raw' format explicitly to remove the restrictions.
WARNING: Image format was not specified for '/mnt/zroot-133/_OS/Android/rpi4/vendor.img' and probing guessed raw.
         Automatically detecting the format is dangerous for raw images, write operations on block 0 will be restricted.
         Specify the 'raw' format explicitly to remove the restrictions.
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.1.93-ge066a024bf41-v8 (build-user@build-host) (Android (10087095, +pgo, +bolt, +lto, -mlgo, based on r487747c) clang version 17.0.2 (https://android.googlesource.com/toolchain/llvm-project d9f89f4d16663d5012e5c09495f3b30ece3d2362), LLD 17.0.2) #1 SMP PREEMPT Tue Aug 13 15:50:22 UTC 2024
[    0.000000] random: crng init done
[    0.000000] Machine model: linux,dummy-virt
[    0.000000] efi: UEFI not found.
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x00000000bfffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000bfffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x00000000bfffffff]
[    0.000000] cma: Reserved 8 MiB at 0x00000000bd000000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: Trusted OS migration not required
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] percpu: Embedded 30 pages/cpu s84392 r8192 d30296 u122880
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 1742098
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 516096
[... 257 lines cut ...]
The error that I get is interesting. It seems that some important file is missing inside the /sys folder of the system.img file ? 
Let's see the content of that folder :

```

root@Z390-AORUS-PRO-DEST:/mnt/zroot-133/_OS/Android/rpi4# mkdir system-rw

root@Z390-AORUS-PRO-DEST:/mnt/zroot-133/_OS/Android/rpi4# sudo mount system.img system-rw

mount: /mnt/zroot-133/_OS/Android/rpi4/system-rw: WARNING: source write-protected, mounted read-only.

root@Z390-AORUS-PRO-DEST:/mnt/zroot-133/_OS/Android/rpi4# cd system-rw

root@Z390-AORUS-PRO-DEST:/mnt/zroot-133/_OS/Android/rpi4/system-rw# ls

acct            bugreports  data           etc              lost+found  oem          sdcard                  system       vendor
apex            cache       data_mirror    init             mnt         postinstall  second_stage_resources  system_dlkm  vendor_dlkm
bin             config      debug_ramdisk  init.environ.rc  odm         proc         storage                 system_ext
bootstrap-apex  d           dev            linkerconfig     odm_dlkm    product      sys                     tmp

root@Z390-AORUS-PRO-DEST:/mnt/zroot-133/_OS/Android/rpi4/system-rw# cd sys
root@Z390-AORUS-PRO-DEST:/mnt/zroot-133/_OS/Android/rpi4/system-rw/sys# ls
```

it's empty. Is normal ? In addition,following your instructions I see that the file userdata.img hasn't been produced. Is this also normal ?

KonstaT, 2024-09-03:
This is not supported in any way at least and can't find any reason why you would need to attempt something like this.

If you need to run AOSP in some sort of emulator, there's an official support way to achieve that.
https://source.android.com/docs/setup/create/avd

Marietto2008, 2024-09-07:
I know but to be able to do it is very intriguing for me.

KonstaT, 2024-09-07:
> I know but to be able to do it is very intriguing for me.

Then I'm not going to spoil the fun from you figuring out.

Your cmdline looks very incorrect at least.
```
...
        -append "console=ttyAMA0,38400 drm.debug=0x0 rootwait rootdelay=5 root=/dev/mmcblk1p2 androidboot.hardware=rpi4 androidboot.selinux=permissive security=selinux selinux=1 androidboot.qemu.hw.mainkeys=0 androidboot.lcd.density=160"
```
On Android rootfs is on ramdisk loaded from ramdisk.img vs. being on any partition and you probably can't have SELinux disabled and enabled at the same time.

Marietto2008, 2024-09-07:
Hello.

I'm not able to find the file userdata.img,but from the messages that I get : 

```
root@Z390-AORUS-PRO-DEST:/mnt/zroot-133/_OS/Android/raspberry-vanilla# ./rpi4-mkimg.sh

Creating image file /mnt/zroot-133/_OS/Android/raspberry-vanilla/out/target/product/rpi4/RaspberryVanillaAOSP14-20240907-rpi4.img...
7168+0 records in
7168+0 records out
7516192768 bytes (7.5 GB, 7.0 GiB) copied, 1.4751 s, 5.1 GB/s
Creating partitions...

Welcome to fdisk (util-linux 2.39.3).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Device does not contain a recognized partition table.
Created a new DOS (MBR) disklabel with disk identifier 0xce1e1fb2.

Comando (m per richiamare la guida): Created a new DOS (MBR) disklabel with disk identifier 0xfb3cd69c.

Comando (m per richiamare la guida): Partition type
   p   primary (0 primary, 0 extended, 4 free)
   e   extended (container for logical partitions)
Select (default p): Numero della partizione (1-4, default 1): First sector (2048-14680063, default 2048): Last sector, +/-sectors or +/-size{K,M,G,T,P} (2048-14680063, default 14680063): 
Created a new partition 1 of type 'Linux' and of size 128 MiB.

Comando (m per richiamare la guida): Partition type
   p   primary (1 primary, 0 extended, 3 free)
   e   extended (container for logical partitions)
Select (default p): Numero della partizione (2-4, default 2): First sector (264192-14680063, default 264192): Last sector, +/-sectors or +/-size{K,M,G,T,P} (264192-14680063, default 14680063): 
Created a new partition 2 of type 'Linux' and of size 2 GiB.

Comando (m per richiamare la guida): Partition type
   p   primary (2 primary, 0 extended, 2 free)
   e   extended (container for logical partitions)
Select (default p): Numero della partizione (3,4, default 3): First sector (4458496-14680063, default 4458496): Last sector, +/-sectors or +/-size{K,M,G,T,P} (4458496-14680063, default 14680063): 
Created a new partition 3 of type 'Linux' and of size 256 MiB.

Comando (m per richiamare la guida): Partition type
   p   primary (3 primary, 0 extended, 1 free)
   e   extended (container for logical partitions)
Select (default e): 
Selected partition 4
First sector (4982784-14680063, default 4982784): Last sector, +/-sectors or +/-size{K,M,G,T,P} (4982784-14680063, default 14680063): 
Created a new partition 4 of type 'Linux' and of size 4.6 GiB.

Comando (m per richiamare la guida): Numero della partizione (1-4, default 4): Hex code or alias (type L to list all): 
Changed type of partition 'Linux' to 'W95 FAT32 (LBA)'.

Comando (m per richiamare la guida): Numero della partizione (1-4, default 4): 
The bootable flag on partition 1 is enabled now.

Comando (m per richiamare la guida): The partition table has been altered.
Syncing disks.

Image mounted as /dev/loop0
Copying boot...
128+0 records in
128+0 records out
134217728 bytes (134 MB, 128 MiB) copied, 1.16107 s, 116 MB/s
Copying system...
2048+0 records in
2048+0 records out
2147483648 bytes (2.1 GB, 2.0 GiB) copied, 55.5621 s, 38.7 MB/s
Copying vendor...
256+0 records in
256+0 records out
268435456 bytes (268 MB, 256 MiB) copied, 5.36113 s, 50.1 MB/s
Creating userdata...
mke2fs 1.47.0 (5-Feb-2023)
Discarding device blocks: fatto                            
Creazione del file system con 1212160 4k blocchi e 303104 inode
Etichetta del file system=ce0805f8-43d1-4aa0-ad2c-be9ad72822a1
Backup del superblocco salvati nei blocchi: 
        32768, 98304, 163840, 229376, 294912, 819200, 884736

Allocating group tables: fatto                            
Scrittura delle tavole degli inode: fatto                            
Creating journal (16384 blocks): fatto
Scrittura delle informazioni dei super-blocchi e dell'accounting del file system: fatto

Done, created /mnt/zroot-133/_OS/Android/raspberry-vanilla/out/target/product/rpi4/RaspberryVanillaAOSP14-20240907-rpi4.img!
```

It seems that userdata.img is called RaspberryVanillaAOSP14-20240907-rpi4.img. Is correct ? I don't see any userdata.img file inside this folder :

/mnt/zroot-133/_OS/Android/raspberry-vanilla/out/target/product/rpi4/

Marietto2008, 2024-09-07:
> Your cmdline looks very incorrect at least.

>-append "console=ttyAMA0,38400 drm.debug=0x0 rootwait rootdelay=5 root=/dev/mmcblk1p2 >androidboot.hardware=rpi4 androidboot.selinux=permissive security=selinux selinux=1 >androidboot.qemu.hw.mainkeys=0 androidboot.lcd.density=160"

This is not VERY incorrect. I'm using that line when I boot Android 10 adapted by Antmicro on my Jetson Nano. You can see the qemu parameters that should be used going here :

https://github.com/antmicro/kvm-aosp-jetson-nano/blob/master/README_Jetson_Nano_BSP.md

At least for Android 10 it works perfectly.

KonstaT, 2024-09-07:
> It seems that userdata.img could be called "RaspberryVanillaAOSP14-20240907-rpi4.img". Is it correct ?

No. Flashable image that the mkimg.sh script creates contains four partitions (boot, system, vendor, and userdata). I even linked the exact line where the userdata partition is created with mkfs.ext4.

Marietto2008, 2024-09-07:
--> I even linked the exact line where the userdata partition is created with mkfs.ext4.

It seems thats  not enough to link "the exact line where the userdata partition is created with mkfs.ext4" if I  don't have that file because it is NOT created for some reason. If your script does not create userdata.img it means that we should at least understand the reasons. The better method is to divide the bigger task into smaller ones to understand where it fails.

Marietto2008, 2024-09-07:
Your script is broken. Something does not work :

```
root@Z390-AORUS-PRO-DEST:/mnt/zroot-133/_OS/Android/raspberry-vanilla# ls /dev/mapper
control  loop0p1  loop0p2  loop0p3  loop0p4

root@Z390-AORUS-PRO-DEST:/mnt/zroot-133/_OS/Android/raspberry-vanilla# sudo mkfs.ext4 /dev/mapper/loop0p4 -I 512 -L userdata

mke2fs 1.47.0 (5-Feb-2023)
Discarding device blocks: fatto                            
Creazione del file system con 1212160 4k blocchi e 303104 inode
Etichetta del file system=f8ea2c6a-c914-4963-92ea-431dbd5fda4d
Backup del superblocco salvati nei blocchi: 
        32768, 98304, 163840, 229376, 294912, 819200, 884736

Allocating group tables: fatto                            
Scrittura delle tavole degli inode: fatto                            
Creating journal (16384 blocks): fatto
Scrittura delle informazioni dei super-blocchi e dell'accounting del file system: fatto

root@Z390-AORUS-PRO-DEST:/mnt/zroot-133/_OS/Android/raspberry-vanilla# ls
nothing is created.
```

KonstaT, 2024-09-07:
> If your script does not create userdata.img it means that we should at least understand the reasons.

FFS, it's not supposed to create any userdata.img, but an userdata partition withing the flashable image. I understand perfectly what it does and why.

> Your script is broken.

No, you just don't understand what it is supposed to do.

As mentioned earlier, what you're attempting is not supported in any shape or form so I don't have time/energy/interest to get into this any further.

If you need to run AOSP in an emulator, use the official Android emulator.
https://source.android.com/docs/setup/create/avd

Marietto2008, 2024-09-07:
> FFS, it's not supposed to create any userdata.img, but an userdata partition within the flashable image. I understand perfectly what it does and why.

So,maybe you don't emulate Android so often,because usually with qemu the userdata partition is declared with :

**-drive index=2,if=none,id=userdata,file=aosp/userdata.img**

But it's not a problem,because now that I've understood where it is,I can use it in qemu doing something like this :

`dd if=/dev/mapper/loop0p4 of=userdata.img`

> If you need to run AOSP in an emulator, use the official Android emulator.

no,sorry. I want to emulate this version of Android.

Marietto2008, 2024-09-08:
One only last question,please. Can you tell me if the content of the userdata.img file that you see below,is correct ? thanks.

[image]