raspberry-vanilla/android_local_manifest #122: Luminosity Control on Custom Android TV ROM 13 for Raspberry Pi 4

rania-taabouri, 2024-09-02:
I am having trouble controlling the screen luminosity on my custom Android TV ROM 13 running on a Raspberry Pi 4. Unlike on my phone, where I can adjust luminosity freely, the Android TV ROM does not include a built-in toolbar or controls for adjusting brightness. Attempts to obtain a brightness control tool from APKPure have been unsuccessful.

KonstaT, 2024-09-03:
Because it's an external display connected via HDMI. It's not technically possible (while I know DDC/CI exists).

closed 2024-09-03 by KonstaT

rania-taabouri, 2024-09-04:
it's the raspberrypi 7inch touch screen what if i switch the build to tablet or automotive will be possibel to control it through a notification bar

KonstaT, 2024-09-04:
Yes, it works on regular tablet target given you've followed https://github.com/raspberry-vanilla/android_local_manifest/wiki/DSI-display to use the official Raspberry 7" DSI display.

Can't recall if Android Automotive has a brightness slider. Android TV does not because you generally can't adjust brightness on an external display.

rania-taabouri, 2024-09-04:
is it possible to add a brightness slider on a the tv target or the automotive if it doesn't exist