raspberry-vanilla/android_local_manifest #123: switching build to automotive

rania-taabouri, 2024-09-04:
i am trying to switch build to automotive by creating a new lunch choice rpi4_car
i tried to combine between rpi4.mk and car.mk to create a rpi4_car.mk withing the arpi/rpi4 folder 
i created a device.mk file :
`

$(call inherit-product, $(SRC_TARGET_DIR)/product/core_64_bit_only.mk)

PRODUCT_NAME := rpi4
PRODUCT_DEVICE := rpi4
PRODUCT_BRAND := arpi
PRODUCT_MANUFACTURER := ARPi
PRODUCT_MODEL := Raspberry Pi 4

include frameworks/native/build/tablet-10in-xhdpi-2048-dalvik-heap.mk

PRODUCT_PROPERTY_OVERRIDES += \
    debug.drm.mode.force=800x480 \
    gralloc.drm.kms=/dev/dri/card0 \
    ro.opengles.version=196609 \
    ro.hardware.vulkan=broadcom \
    debug.stagefright.c2-poolmask=0x350000 \
    ro.vendor.v4l2_codec2.decode_concurrent_instances=4 \
    wifi.interface=wlan0 \
    ro.rfkilldisabled=1

PRODUCT_SOONG_NAMESPACES += external/v4l2_codec2

PRODUCT_PACKAGES += \
    RpFrameworkOverlay

PRODUCT_PACKAGES += \
    gralloc.rpi4 \
    vulkan.broadcom \
    memtrack.rpi4 \
    audio.primary.rpi4 \
    audio.usb.default \
    audio.r_submix.default \
    libc2plugin_store \
    wificond \
    wifilogd \
    wpa_supplicant \
    wpa_supplicant.conf \
    hostapd \
    libbt-vendor

PRODUCT_PACKAGES += \
    libEGL_mesa \
    libGLESv1_CM_mesa \
    libGLESv2_mesa \
    libgallium_dri \
    libglapi

PRODUCT_PACKAGES += \
    android.hardware.graphics.allocator@2.0-service.rpi4 \
    android.hardware.graphics.mapper@2.0-impl.rpi4 \
    android.hardware.graphics.composer@2.1-service.rpi4 \
    android.hardware.camera.provider@2.5-external-service \
    android.hardware.audio@4.0-impl \
    android.hardware.audio.effect@4.0-impl \
    android.hardware.audio.service \
    android.hardware.keymaster@3.0-impl \
    android.hardware.keymaster@3.0-service \
    android.hardware.gatekeeper@1.0-service.software \
    android.hardware.memtrack@1.0-impl \
    android.hardware.memtrack@1.0-service \
    android.hardware.health@2.1-service \
    android.hardware.health@2.1-impl \
    android.hardware.health.storage@1.0-service \
    android.hardware.wifi@1.0-service \
    android.hardware.bluetooth@1.0-service \
[... 108 lines cut ...]
    android.hardware.broadcastradio-service.default

PRODUCT_COPY_FILES += \
    frameworks/native/data/etc/android.hardware.broadcastradio.xml:$(TARGET_COPY_OUT_VENDOR)/etc/permissions/android.hardware.broadcastradio.xml

ENABLE_CAMERA_SERVICE := true

ENABLE_CAREVSSERVICE_SAMPLE := true
ENABLE_EVS_SAMPLE := true
ENABLE_EVS_SERVICE := true
ENABLE_REAR_VIEW_CAMERA_SAMPLE := true

PRODUCT_COPY_FILES += \
    frameworks/native/data/etc/android.software.activities_on_secondary_displays.xml:$(TARGET_COPY_OUT_VENDOR)/etc/permissions/android.software.activities_on_secondary_displays.xml \
    frameworks/native/data/etc/car_core_hardware.xml:$(TARGET_COPY_OUT_VENDOR)/etc/permissions/car_core_hardware.xml

# Vehicle HAL configuration
PRODUCT_PACKAGES += \
    android.hardware.automotive.vehicle@2.0-default-service
    
   `
```
and made the necessary changes in AndroidProduct.mk file 

the rpi4_car-eng lunch choice was successfully added but i am facing a lot of build issues

KonstaT, 2024-09-04:
Raspberry Vanilla project already supports Android Automotive build target. There's no need to add anything. Please see the build instructions (though https://github.com/raspberry-vanilla/android_local_manifest/issues/83 is something to consider).

rania-taabouri, 2024-09-04:
i know sir i just want to learn how to create a new one myself it's my first time with aosp i didn't know about vanilla project so i worked with another repo that doesn't have an automotive build only tv one that's why i am tying to do it myself and it's a chance to learn i really appreciate your help