raspberry-vanilla/android_local_manifest #124: lunch aosp_rpi5-ap3a-userdebug failed immediatly

lohriialo, 2024-09-09:
I've got a new pi5 I want to try and build aosp and ran into the following error

- ```repo sync``` was successful
- ```. build/envsetup.sh``` no error
- ```lunch aosp_rpi5-ap3a-userdebug``` results in the following error and quit immediately
```
============================================
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=15
TARGET_PRODUCT=aosp_rpi5
TARGET_BUILD_VARIANT=userdebug
TARGET_ARCH=arm64
TARGET_ARCH_VARIANT=armv8-a
TARGET_CPU_VARIANT=cortex-a76
TARGET_2ND_ARCH=arm
TARGET_2ND_ARCH_VARIANT=armv7-a-neon
TARGET_2ND_CPU_VARIANT=generic
HOST_OS=linux
HOST_OS_EXTRA=Linux-6.8.0-41-generic-x86_64-Ubuntu-24.04.1-LTS
HOST_CROSS_OS=windows
BUILD_ID=AP3A.240905.015.A2
OUT_DIR=out
============================================

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  Wondering whether to use user, userdebug or eng?

  user        The builds that ship to users. Reduced debugability.
  userdebug   High fidelity to user builds but with some debugging options
              enabled. Best suited for performance testing or day-to-day use
              with debugging enabled.
  eng         More debugging options enabled and faster build times, but
              runtime performance tradeoffs. Best suited for day-to-day
              local development when not doing performance testing.
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

KonstaT, 2024-09-09:
There's no error and that's exactly what's supposed to happen.

closed 2024-09-09 by KonstaT

lohriialo, 2024-09-09:
@KonstaT does that mean it's currently not possible to compile following the instructions [here] (https://github.com/raspberry-vanilla/android_local_manifest/blob/android-15.0/README.md)? appreciate a short explaination

KonstaT, 2024-09-09:
How would it indicate that?

You've done step 6 (select device and build target), step 7 & 8 to go (compile & make flashable image).

lohriialo, 2024-09-09:
@KonstaT thank you, my bad!