raspberry-vanilla/android_local_manifest #127: AAOS RPi4 - BT Call initiated but no HFP audio on device

nimesh-d, 2024-09-18:
Device -  RPi4 8GB

When making a phone call from connected cell phone or from recent calls screen in AAOS phone app,  call intiates and connects successfully.
But there is no HFP audio in RPi HDMI outs. 
Where as when playing media (like youtube) from phone, the A2DP audio works fine and I can here media audio from RPi HDMI output.

Following profiles are enabled as recommended except `map.client` ( i had to disable map due to crashes which seems unrelated),

`# The Bluetooth profiles that cars expect to have enabled. All other profiles
# are disabled by default.
bluetooth.profile.a2dp.sink.enabled=true
bluetooth.profile.avrcp.controller.enabled=true
bluetooth.profile.gatt.enabled=true
bluetooth.profile.hfp.hf.enabled=true
bluetooth.profile.map.client.enabled=false
bluetooth.profile.pan.nap.enabled=true
bluetooth.profile.pan.panu.enabled=true
bluetooth.profile.pbap.client.enabled=true
`

We have meen stuck on this issue for months and would really appreciate any help to resolve the issue.
cc @KonstaT

nimesh-d, 2024-09-18:
Call audio issue is present in Android-15 branch as well.

Also noticed the media audio paused immediately after starting. (This must be related to a2dp_sink_automatically_request_audio_focus property).
cc @KonstaT

KonstaT, 2024-09-19:
> When making a phone call from connected cell phone or from recent calls screen in AAOS phone app, call intiates and connects successfully.
> But there is no HFP audio in RPi HDMI outs.
> Where as when [...]

This has never worked. AFAIK Android bluetooth stack doesn't support SCO over HCI (vs. bluez stack on regular Linux distros). Also bluetooth hardware in Pi is not connected via PCM/I2S which I beleive is quite commonly used for HFP.

Further reading (quite old but I don't think anything relevant has changed):
https://discuss.96boards.org/t/hfp-client-sco-audio-routing/3532/121
https://groups.google.com/g/btstack-dev/c/pZmp9QVWYBk
https://forums.raspberrypi.com/viewtopic.php?t=206970

> Also noticed the media audio paused immediately after starting.

Could be related to the commit I made following upstream configuration. It seems odd that both ADB source and sink would be enabled at the same. Likely defaults to source. You're welcome to revert the change in rpi4 device tree to see if makes a difference.
https://github.com/raspberry-vanilla/android_device_brcm_rpi4/commit/741b658768af9b48058c9d6d2c32ade4bd158b98
https://android.googlesource.com/device/google_car/+/61473cb0ebd9e6071fa9e219be43651337fae752
https://android.googlesource.com/platform/packages/services/Car/+/c126f60c1bb6f0a557acecfefa6f4b6a54c10ead

closed 2024-09-19 by KonstaT

nimesh-d, 2024-09-19:
> > When making a phone call from connected cell phone or from recent calls screen in AAOS phone app, call intiates and connects successfully.
> > But there is no HFP audio in RPi HDMI outs.
> > Where as when playing media (like youtube) from phone, the A2DP audio works fine and I can here media audio from RPi HDMI output.
> 
> This has never worked. AFAIK Android bluetooth stack doesn't support SCO over HCI (vs. bluez stack on regular Linux distros). Also bluetooth hardware on Pi is not connected via PCM/I2S which I believe is quite commonly used for HFP.
> 
> Further reading (quite old but I don't think anything relevant has changed): https://discuss.96boards.org/t/hfp-client-sco-audio-routing/3532/121 https://groups.google.com/g/btstack-dev/c/pZmp9QVWYBk https://forums.raspberrypi.com/viewtopic.php?t=206970
> 
> > Also noticed the media audio paused immediately after starting.
> 
> Could be related to the commit I made following upstream configuration. It seems odd that both A2DB source and sink would be enabled at the same. Likely defaults to source. You're welcome to revert the change in rpi4 device tree to see if it makes a difference. [raspberry-vanilla/android_device_brcm_rpi4@741b658](https://github.com/raspberry-vanilla/android_device_brcm_rpi4/commit/741b658768af9b48058c9d6d2c32ade4bd158b98) https://android.googlesource.com/device/google_car/+/61473cb0ebd9e6071fa9e219be43651337fae752 https://android.googlesource.com/platform/packages/services/Car/+/c126f60c1bb6f0a557acecfefa6f4b6a54c10ead

Thank You!