raspberry-vanilla/android_local_manifest #129: No sound output on rpi5 with AAOS

liquanzeng-calibur, 2024-09-21:
AAOS Version:
`repo init -u https://android.googlesource.com/platform/manifest -b android-14.0.0_r67`

No deivce found in ALSA
rpi5:/ $ alsa_aplay -l                                                                                                                                                                                            
aplay: device_list:279: no soundcards found...
rpi5:/ $ alsa_aplay -L                                                                                                                                                                                            
null
    Discard all samples (playback) or generate zero samples (capture)

USB speecker connected;

adb shell cat /proc/asound/cards
 0 [vc4hdmi0       ]: vc4-hdmi - vc4-hdmi-0
                      vc4-hdmi-0
 1 [vc4hdmi1       ]: vc4-hdmi - vc4-hdmi-1
                      vc4-hdmi-1
 2 [USB            ]: USB-Audio - Jabra SPEAK 510 USB
                      Jabra SPEAK 510 USB at usb-xhci-hcd.0-1, full speed

Force close when enter sound setting in system setting.
No sound output when play a sound, but mic work normally.

So need modify some setting for AAOS ?
Normally worked with lunch aosp_rpi5-ap2a-userdebug.

KonstaT, 2024-09-23:
You would need to have root access for `alsa_aplay` to be able to list anything. As you can see from `cat /proc/asound/cards`, ALSA devices are present as expected.

I recently did Android Automotive builds from `android-14.0.0_r67` and `android-15.0.0_r1` branches. I also noted that sound settings option crashes on AOSP14 QPR3 Android Automotive. IIRC audio output via HDMI still worked. It's an issue with upstream AOSP and Android Automotive targets have been different level broken since AOSP 14 QPR1 (see e.g. https://github.com/raspberry-vanilla/android_local_manifest/issues/83). It's not something can/should be fixed in the Raspberry Pi devices specific implementation that this project provides. There was no issue with sound settings nor audio output on `android-15.0.0_r1`.

So, as I see you have three options.
1. Use AOSP 14 QPR1 (i.e. `android-14.0.0_r22` - https://github.com/raspberry-vanilla/android_local_manifest/tree/android-14.0.0_r22)
2. Fix it on AOSP 14 QPR 3 (i.e. `android-14.0.0_r67`)
3. Use AOSP 15 (i.e. `android-15.0.0_r1`)

liquanzeng-calibur, 2024-09-25:
> You would need to have root access for `alsa_aplay` to be able to list anything. As you can see from `cat /proc/asound/cards`, ALSA devices are present as expected.
> 
> I recently did Android Automotive builds from `android-14.0.0_r67` and `android-15.0.0_r1` branches. I also noted that sound settings option crashes on AOSP14 QPR3 Android Automotive. IIRC audio output via HDMI still worked. It's an issue with upstream AOSP and Android Automotive targets have been different level of broken since AOSP14 QPR1 (see e.g. #83). It's not something that can be fixed in the Raspberry Pi devices specific implementation that this project provides. There was no issue with sound settings nor audio output on `android-15.0.0_r1` Android Automotive.
> 
> So, as I see you have three options.
> 
> 1. Use AOSP14 QPR1 (i.e. `android-14.0.0_r22` - https://github.com/raspberry-vanilla/android_local_manifest/tree/android-14.0.0_r22)
> 2. Fix it on AOSP14 QPR3 (i.e. `android-14.0.0_r67`)
> 3. Use AOSP15 (i.e. `android-15.0.0_r1`)

Thanks for your help, Sound setting doesn't force close on android-15.0.0_r1 branche, but volume setting doesn't work, awalys keep same volume even after I modified the setting.

KonstaT, 2024-09-25:
On `android-15.0.0_r1` adjusting the volume level with volume keys works at least.

I think Android Automotive build target is mainly tested by Google on devices with fixed volume on the OS as generally you could probably expect a car to have an amplifier.
https://android.googlesource.com/device/google/cuttlefish/+/refs/tags/android-15.0.0_r1/shared/auto/overlay/frameworks/base/core/res/res/values/config.xml#32

closed 2024-09-26 by liquanzeng-calibur