raspberry-vanilla/android_local_manifest #133: Android not booting up

vishak-kurup, 2024-10-03:
I could successfully make the image. I flashed it to the SD card. But it is not booting. In the display, it shows the error "Firmware not found". I am building android15 for raspberyypi5.

KonstaT, 2024-10-05:
But are you trying to boot the image on a Raspberry Pi 5? This error message would seem strange on Pi 5 where the firmware is part of EEPROM (i.e. it's not possible to not have firmware) vs. Pi 4 where the boot firmware is part of the OS (i.e. on the boot partition).

In any case, difficult to say without seeing the actual error message, logs, and what the image you've created actually looks like.

vishak-kurup, 2024-10-07:
It was a big mistake from my side. My board is rpi-4 not rpi-5.
I am now building the image for rpi-4.

closed 2024-10-07 by vishak-kurup