raspberry-vanilla/android_local_manifest #134: Build Failure with /bin/bash: line 1: meson: command not found

Viralblack, 2024-10-06:
Followed both build guides for AOSP and the additional installation instructions, configured a new Ubuntu 24 explicitly for this purpose. 
[image]
Following that trail it appeared meson is incorrectly configured, attempted a few different ways to correctly configure it but all seemed to point in the same direction. When running the installs for the environment given, this warning appears.
[image]
Attempted a few routes to resolve, didnt seemingly change anything. Potentially more environment/user error, just unsure what direction to go in.

Viralblack, 2024-10-06:
The error in installation also takes place if using Debian 12 as base, it appears on the second line of step 2 with the python modules. 
[image]

Viralblack, 2024-10-06:
From the second line of the installation of pip3 elements, I used pipx to install them and jinja, pyyaml, and dataclasses post the same error of it being an externally managed environment. Will still continue to see if pipx remediates the compile error

Viralblack, 2024-10-06:
Wider and more prolific failures across the board with Debian as opposed to Ubuntu
[image]

Viralblack, 2024-10-07:
Made forward progress, rebuilt the environment (after double checking Ubuntu is the recommended build environment for AOSP and triple checking their requirements) and worked to remediate the MESON dependency first. I followed a few of the recommended solutions, pipx did not work for me, but venv did appear to allow for the implementation of MESON, so I pip installed all the sources from the earlier step in the virtual environment for pip and continued the build. It ran, but then said MESON was disallowed.

[image]

The error cited the path tools section as why MESON cant be used, per its statements. 

[image]

Will be attempting to see if I can set the workaround variable.

Viralblack, 2024-10-07:
Steps I followed to circumvent externally managed warning outlined here https://stackoverflow.com/questions/75608323/how-do-i-solve-error-externally-managed-environment-every-time-i-use-pip-3

Viralblack, 2024-10-07:
Adding the tool to the soong config allowed MESON to run, but its now complaining about MAKO, another second level build tool added by this build's requirements specifically. 

[image]

Viralblack, 2024-10-09:
> hi, I have faced the same issue. I don't think it because of any meson build issue. This is because of some memory issue. just give the below commands before your make command. surely this will solve the issue. better to do make clean once (if your build is corrupted)
> 
> sudo swapoff -a sudo dd if=/dev/zero of=/swapfile bs=1G count=32 sudo mkswap /swapfile sudo swapon /swapfile grep SwapTotal /proc/meminfo

Gave it a shot this morning, with my setup that occupied 100% of the drive no matter how much I expanded the partition. I additionally tried setting the build environment to 64gb from 32gb and it didnt change, for better or worse.

sajkhan1988, 2024-10-17:
Got the same problem. Has anyone managed to solve it ?

vnphuong16, 2024-10-17:
I encountered the same issue. I tried using --break-system-packages with pip3 to install the component.
Then, I attempted to disable AppArmor, but it still didn't work. Finally, I downgraded to Ubuntu 22.04 LTS, and the build worked as expected.
(https://bugs.launchpad.net/ubuntu/+source/apparmor/+bug/2063976)

sajkhan1988, 2024-10-17:
Ill downgrade and give it a go as well. Thanks.

sajkhan1988, 2024-10-17:
I tried building in Ubuntu 22.04 LTS and builds fails around 70% of completion. I don't see any particular error or failure reason in the logs. Logs attached. 
@vnphuong16  Am I missing something?  Space is not an issue, I got like 263 GB still free. I am building this in WSL though. Any advice? Thanks. 
[buildlogs.txt](https://github.com/user-attachments/files/17421607/buildlogs.txt)

MagicAndre1981, 2024-10-17:
I also got this in Ubuntu 24.04 and reverted to 22.04 and it worked.

In your log I see this:

`19:56:22 ninja failed with: exit status 137`

Error 137 is low memory issue

> This error code means that your process have been killed due to your machine doesn't have enough memory.

So make sure RAM + swap is 64GB. During compile I see memory usgae over 50GB for several steps while normal usage is between 12 and 20GB. I used [MissionCenter a Windows Taskmgr replica](https://flathub.org/apps/io.missioncenter.MissionCenter) to see the memory usage during build.

[image]

Viralblack, 2024-10-18:
Attempted the build with the recommended Ubuntu 22 as suggested, seemed to build further but unfortunately died when it started doing some of the auto specific parts of the build. My build target is RPi5-auto and that could be different from some.
[image]

Has anyone had success with Rpi5-Auto target working and building? I love the idea of running this on a setup I've built for a head unit for a friends offroad rig, but I cant seem to get it to build to save my soul. If anyone is having success feel free to shoot me a message, I'd happily pay for a working download flashable for Rpi5-Auto.

MagicAndre1981, 2024-10-18:
This week I build RPi4 ATV and it worked for me.

Around 50% there is a place where memory usage grows a lot. How much RAM do you have?

sajkhan1988, 2024-10-18:
> I also got this in Ubuntu 24.04 and reverted to 22.04 and it worked.
>
> In your log I see this:
>
> `19:56:22 ninja failed with: exit status 137`
>
> Error 137 is low memory issue
>
>
> So make sure RAM + swap [...]

Yes I think that could be the issue. I did have some heavy processes running along side the build. I got 64GB RAM. I will try to run the build without any other processes running in the system.

sajkhan1988, 2024-10-18:
> This week I build RPi4 ATV and it worked for me.
> 
> Around 50% there is a place where memory usage grows a lot. How much RAM do you have?

Makes sense. I got 64GB RAM but I had some other processes running which might have caused this error. Will run the build without any other application running.

KonstaT, 2024-10-22:
Build should work Ubuntu 22.04 LTS on a machine that meets the hardware requirements.
https://source.android.com/docs/setup/start/requirements

I've updated build instructions to include this information.
https://github.com/raspberry-vanilla/android_local_manifest/commit/f3dfe83910a8a42b9048d1272604f73adb53eb92

closed 2024-10-22 by KonstaT