raspberry-vanilla/android_local_manifest #135: CAN network state of art

demoa00, 2024-10-10:
I want to know the state of the implementation of the CAN network for the configuration rpi4 with Android Automotive OS 14. In the `aosp_rpi4_car.mk` there is a section about the CAN network to include its functionalities, from the system log I see that the `CanController` start but I don't see any interface for the CAN (e.g. can0) when in the ADB shell i use the command `ip link show`. There are some other changes to do in order to make up the CAN network? Sorry in advance if I wrote in the wrong place. Thanks.

KonstaT, 2024-10-22:
Goal of the Raspberry Vanilla project is to support the built-in hardware on Raspberry Pi. There's no support for external CAN modules.

You would need to compile the kernel with `CAN` enabled (https://github.com/raspberry-vanilla/android_kernel_brcm_rpi/blob/android-14.0/arch/arm64/configs/android_rpi4_defconfig#L1461) and whatever other drivers your hardware may require for the can device to appear.

You would also need to see what modifications the default AOSP CAN HAL require to work on that specific hardware.
https://android.googlesource.com/platform/hardware/interfaces/+/refs/tags/android-14.0.0_r67/automotive/can/aidl/default/

closed 2024-10-22 by KonstaT

demoa00, 2024-10-22:
Thanks for the answer. I have request the state of CAN network because i seen that at the path `device/brcm/rpi4-kernel/overlays` some overlays related to the CAN network such us `waveshare-can-fd-hat-mode-a.dtbo` and i supposed that this was part of the Raspberry Vanilla project. So only for avoid other dubs, what is the code related to the Vanilla Project and the code that is part of the default Raspberry implementation?
Thanks.

KonstaT, 2024-10-22:
Looking at that specific overlay (https://github.com/raspberry-vanilla/android_kernel_brcm_rpi/blob/android-14.0/arch/arm/boot/dts/overlays/waveshare-can-fd-hat-mode-a-overlay.dts) you would need to compile the kernel (https://github.com/raspberry-vanilla/android_kernel_manifest/tree/android-14.0) with `CAN` and `CAN_MCP251XFD` enabled to use it.

All the overlays that Pi Foundation's Linux kernel fork includes, are also included with my Android kernel. It still doesn't mean that every driver to use every overlays is built into the Android kernel. You can consider Android an OS for embedded devices and it's designed to support set of specific hardware at a time (vs. your regular Linux distribution).