raspberry-vanilla/android_local_manifest #139: rpi5 build error

lohriialo, 2024-10-23:
Didn't modify anything, just stock build on this repo, ran into following build error
```
FAILED: 
external/ffmpeg/libavcodec/Android.mk: error: "libavcodec (SHARED_LIBRARIES android-arm64) missing libdav1d (SHARED_LIBRARIES android-arm64)" 
Available variants:
  libdav1d (STATIC_LIBRARIES linux-x86_64)
  libdav1d (STATIC_LIBRARIES linux-x86)
  libdav1d (STATIC_LIBRARIES android-arm64)
  libdav1d (STATIC_LIBRARIES android-arm)
You can set ALLOW_MISSING_DEPENDENCIES=true in your environment if this is intentional, but that may defer real problems until later in the build.
```

KonstaT, 2024-10-23:
There is https://github.com/raspberry-vanilla/android_external_libdav1d/commit/4bd7c179a31db2393e0001ef25de17e824ffcada in Raspberry Vanilla's libdav1d fork because of this. Your `external/libdav1d` is not synced with the project.

Having no insight on steps you've taken to sync the source or even the branch you've used, I can only guess you haven't followed the steps on 'How to build' or ignored errors during `repo sync`. The solution is to sync your source code following the provided documentation.
https://github.com/raspberry-vanilla/android_local_manifest/tree/android-15.0

closed 2024-10-23 by KonstaT

lohriialo, 2024-10-23:
@KonstaT I've followed the steps in [readme](https://github.com/raspberry-vanilla/android_local_manifest/blob/android-15.0/README.md)

after setting up the build environment, here's the steps I followed
step 1
```
repo init -u https://android.googlesource.com/platform/manifest -b android-15.0.0_r2 --depth=1
curl -o .repo/local_manifests/manifest_brcm_rpi.xml -L https://raw.githubusercontent.com/raspberry-vanilla/android_local_manifest/android-15.0/manifest_brcm_rpi.xml --create-dirs
curl -o .repo/local_manifests/remove_projects.xml -L https://raw.githubusercontent.com/raspberry-vanilla/android_local_manifest/android-15.0/remove_projects.xml
```
step 2
`repo sync`
step 3
`. build/envsetup.sh`
step 4
`lunch aosp_rpi5-ap3a-userdebug`
step 4
`make bootimage systemimage vendorimage -j1`

KonstaT, 2024-10-23:
> I can see [libdav1d](https://github.com/raspberry-vanilla/android_external_libdav1d) is present at external/libdav1d

But is it up to date with Raspberry Vanilla (and not the one you'd get from AOSP)? What does `git rev-parse HEAD` return on `external/libdav1d`?

lohriialo, 2024-10-23:
`git rev-parse HEAD` cf95d5eba159be89106ae4455c9f7d6907ee1d0d

KonstaT, 2024-10-23:
So, that's what you would get from AOSP (https://android.googlesource.com/platform/external/libdav1d/+/refs/tags/android-15.0.0_r2), not what you'd get from Raspberry Vanilla i.e. `d7edc76018db3209d83c6a2b996e0df3894a6113` (https://github.com/raspberry-vanilla/android_external_libdav1d/tree/android-15.0).

I can only assume that you have synced AOSP, after that you've added the local manifest, synced again and ignored all the failing repositories (i.e. this project's every AOSP replacement/remove-project in the local manifest https://github.com/raspberry-vanilla/android_local_manifest/blob/android-15.0/manifest_brcm_rpi.xml). Sync the failing repositories with `--force-sync` flag as `repo` has already suggested you or start from the scratch following the instructions.

lohriialo, 2024-10-23:
@KonstaT thank you, just did a --force-sync, hopefully that was it