raspberry-vanilla/android_local_manifest #140: ChipDipDAC

chipdipru, 2024-10-24:
Hi @KonstaT 
It looks like the I2S audio card ChipDipDAC is turned off.
`# CONFIG_SND_BCM2708_SOC_CHIPDIP_DAC is not set  `
Please help me turn on the ChipDipDAC as the main audio output.

KonstaT, 2024-10-24:
You would need to compile the Pi 4/5 kernels (https://github.com/raspberry-vanilla/android_kernel_manifest/tree/android-15.0) with `SND_BCM2708_SOC_CHIPDIP_DAC` enabled (https://github.com/raspberry-vanilla/android_kernel_brcm_rpi/blob/android-15.0/arch/arm64/configs/android_rpi4_defconfig#L4513 & https://github.com/raspberry-vanilla/android_kernel_brcm_rpi/blob/android-15.0/arch/arm64/configs/android_rpi4_defconfig#L4513). I don't have the hardware to test so pull requests are welcome once you've tested this works on the kernel side.

The project's wiki provides documentation on how to change the audio output device on Raspberry Vanilla builds.
https://github.com/raspberry-vanilla/android_local_manifest/wiki/Audio

It seems that the docs haven't been updated in a while and on Pi 5 the ALSA device number for the DAC is likely 2 (i.e. persist.audio.pcm.card=2). You can use e.g. `cat /proc/asound/cards` to verify the cards numbers.

ChipDipDAC, 2024-10-28:
I enabled the kconfig options you mentioned and everything worked with the freshly built kernel, thank you for your help! Could you please include our device in your next builds?

chipdipru, 2024-10-28:
I enabled the config options you mentioned and everything worked with the freshly built kernel, thank you for your help! Could you please include our device in your next builds?
p.s.Sorry, I replied earlier from another account.

KonstaT, 2024-10-29:
I have enabled the kernel configurations and it will included if/when there are future builds.

closed 2024-10-29 by KonstaT