raspberry-vanilla/android_local_manifest #142: mkimg.sh of aosp15 making wrong disk image for rpi4b

arminghani, 2024-10-25:
Hi
I've built aosp 15 for rpi4 with provided instructions set. When I use mkimg.sh script to make disk image, it's successful but when I flash on an sd card the result is that all partitions are curropted. for example when I try to mount first partition on my machine (boot partition) I get error:
```
user@laptop:~$ sudo mount /dev/sda1 /mnt/disk/
mount: /mnt/disk: wrong fs type, bad option, bad superblock on /dev/sda1, missing codepage or helper program, or other error.
```
However the boot.img in out directory is being mounted fine but when it's cloned into disk with mkimg.sh it's not successful.
Regards

KonstaT, 2024-10-26:
What Linux distro are you using?

What is the output of running `./rpi4-mkimg.sh` from the root of the AOSP source tree after you've built the `rpi4` target?

arminghani, 2024-10-26:
the mkimg script in root worked
I was using the one in device directory.
Thanks

closed 2024-10-26 by arminghani