raspberry-vanilla/android_local_manifest #145: sampling rate is limited to 48 kHz

chipdipru, 2024-10-29:
Hi Konsta
When playing audio through a DAC, the sampling rate is limited to 48 kHz.
Is it possible to remove the limitation in Android?

KonstaT, 2024-10-29:
Android audio HAL that is used with external GPIO audio DACs uses 48 kHz sampling rate.
https://github.com/raspberry-vanilla/android_device_brcm_rpi5/blob/android-15.0/audio/audio_hw.c#L56

Primary output is also configured to use 48 kHz as there's different audio output devices/drivers to take into consideration (3.5mm jack, HDMI, DAC).
https://github.com/raspberry-vanilla/android_device_brcm_rpi5/blob/android-15.0/audio/audio_policy_configuration.xml#L22-L26

You're welcome to investigate if you can adjust the sampling rate for your DAC hardware but I doubt it's something that can be merged here.

chipdipru, 2024-11-06:
I definitely don't have the resources to automatically set the sampling rate for the GPIO DAC.
Thousands of owners of similar DACs will be grateful to you if you can do this for us.

KonstaT, 2024-11-07:
As mentioned, default configuration needs to support different audio output devices (3.5mm jack, HDMI, DAC).

closed 2024-11-07 by KonstaT