raspberry-vanilla/android_local_manifest #146: Screen orientation.

SiderM, 2024-10-29:
In the Konsta build, there is a screen rotation option in the RPI settings. How to implement screen rotation in aosp? I didn't find any information about this in the wiki. Tell me please.

SiderM, 2024-10-29:
And another question. Will this work in aosp car?

KonstaT, 2024-10-29:
Closing as this is not an issue with the Raspberry Vanilla project.

You can change the user rotation as you generally would on Android.
https://stackoverflow.com/questions/25864385/changing-android-device-orientation-with-adb
https://cs.android.com/android/platform/superproject/+/android-15.0.0_r2:frameworks/base/packages/SettingsProvider/res/values/defaults.xml?q=def_user_rotation

There's also several third party applications for the purpose.