raspberry-vanilla/android_local_manifest #151: can't open can't open

RubinPro, 2024-11-26:
[ 99% 158183/158567] Target boot debug image: out/target/product/rpi4/boot-debug.img (priority: 6)
FAILED: out/target/product/rpi4/boot-debug.img
/bin/bash -c "out/host/linux-x86/bin/mkbootimg --kernel out/target/product/rpi4/kernel  --ramdisk out/target
/product/rpi4/ramdisk-debug.img --cmdline \"console=ttyS0,115200 no_console_suspend root=/dev/ram0 rootwait 
androidboot.hardware=rpi4 androidboot.selinux=permissive\" --os_version 14 --os_patch_level 2024-05-05  --ou
tput  out/target/product/rpi4/boot-debug.img"
usage: mkbootimg [-h] [--kernel KERNEL] [--ramdisk RAMDISK] [--second SECOND]
                 [--dtb DTB]
                 [--recovery_dtbo RECOVERY_DTBO | --recovery_acpio RECOVERY_ACPIO]
                 [--cmdline CMDLINE] [--vendor_cmdline VENDOR_CMDLINE]
                 [--base BASE] [--kernel_offset KERNEL_OFFSET]
                 [--ramdisk_offset RAMDISK_OFFSET]
                 [--second_offset SECOND_OFFSET] [--dtb_offset DTB_OFFSET]
                 [--os_version OS_VERSION] [--os_patch_level OS_PATCH_LEVEL]
                 [--tags_offset TAGS_OFFSET] [--board BOARD]
                 [--pagesize {2048,4096,8192,16384}] [--id]
                 [--header_version HEADER_VERSION] [-o OUTPUT]
                 [--vendor_boot VENDOR_BOOT] [--vendor_ramdisk VENDOR_RAMDISK]
                 [--vendor_bootconfig VENDOR_BOOTCONFIG]
                 [--gki_signing_algorithm GKI_SIGNING_ALGORITHM]
                 [--gki_signing_key GKI_SIGNING_KEY]
                 [--gki_signing_signature_args GKI_SIGNING_SIGNATURE_ARGS]
                 [--gki_signing_avbtool_path GKI_SIGNING_AVBTOOL_PATH]
mkbootimg: error: argument --kernel: can't open 'out/target/product/rpi4/kernel': [Errno 2] No such file or 
directory: 'out/target/product/rpi4/kernel'
13:34:32 ninja failed with: exit status 1

#### failed to build some targets (02:15:55 (hh:mm:ss)) ####

KonstaT, 2024-11-27:
Yet the kernel get copied in https://github.com/raspberry-vanilla/android_device_brcm_rpi4/blob/android-14.0/device.mk#L201-L203.

Also `mkbootimg` is not used for anything in Raspberry Vanilla project as we're using a custom one.
https://github.com/raspberry-vanilla/android_build/commit/c075bf67ea7fae3231ddbfbba8dab86d2edd87e2
https://github.com/raspberry-vanilla/android_device_brcm_rpi4/blob/android-14.0/BoardConfig.mk#L46-L47
https://github.com/raspberry-vanilla/android_device_brcm_rpi4/blob/android-14.0/mkbootimg.mk

I can only assume you're doing something other than what is instructed in the project's documentation.
https://github.com/raspberry-vanilla/android_local_manifest

closed 2024-11-27 by KonstaT

RubinPro, 2024-12-01:
No, I have built it exactly according to the documentation provided on your GitHub and I still get this error.

source i downloaded
repo init -u https://android.googlesource.com/platform/manifest -b android-14.0.0_r34
curl -o .repo/local_manifests/manifest_brcm_rpi.xml -L https://raw.githubusercontent.com/raspberry-vanilla/android_local_manifest/android-14.0.0_r34/manifest_brcm_rpi.xml --create-dirs

device i lunched
lunch aosp_rpi4-ap1a-userdebug