raspberry-vanilla/android_local_manifest #153: YouTube videos freeze

secuflag, 2024-12-03:
RPi5: Every videos on YouTube freeze for few seconds if I try to go back by tapping on the video.

I see a tons of this message in logcat;
`W SimpleC2Component: unknown frame index: xxxx`
where `xxxx` is a number.

YouTube version: 19.47.41
ROM: AOSP15-20240905-KonstaKANG-rpi5

Any help would be appreciated.

KonstaT, 2024-12-03:
Is this even an issue?

Very close to just closing this because this issue tracker is intended for the Raspberry Vanilla project (not my personal releases). Google Play services required to run Google applications such as YouTube are not part of Raspberry Vanilla project (nor my personal releases).

Please include steps how to reproduce this issue (install which GApps how, tap where what how, etc) and attach appropriate logs (complete `logcat` output instead of just one line).

secuflag, 2024-12-05:
Sorry, I forgot to mention that I have the same issue on Vanilla builds.

Steps to reproduce;
- Install AOSP15-20240905-KonstaKANG-rpi5 or a Vanilla build
- Install GApps by MindTheGapps (https://github.com/MindTheGapps/15.0.0-arm64/releases/download/MindTheGapps-15.0.0-arm64-20240928_150548/MindTheGapps-15.0.0-arm64-20240928_150548.zip)
- Login to your Google account.
- Install YouTube from Google Play Store
- Open any video (try with 4-5 videos) and try to rewind a few seconds by tapping on the video, at this point the video will freeze for some seconds or it plays in very slow frame rate.

Logcat attached.

[logcat.txt](https://github.com/user-attachments/files/18023285/logcat.txt)

KonstaT, 2024-12-12:
Looking at the `logcat`, there are video streams that use AV1 and VP9 codecs with `c2.ffmpeg.av1.decoder` and `c2.ffmpeg.vp9.decoder` decoders, respectively. Maybe this explains the different behaviour (freeze vs. slow frame rate)? Which one is which?

My personal builds offer option to change the video decoders/encoder (Settings -> System -> Raspberry Pi settings -> Hardware video decoding/encoding). Same thing on Raspberry Vanilla is documented on https://github.com/raspberry-vanilla/android_local_manifest/wiki/Video-decoding-&-encoding. On Pi 5 you can maybe get away using Android's own AV1/VP9 software decoders instead of FFmpeg.

secuflag, 2024-12-15:
I have the same issue (on YT app and on Chrome browser) also if I set `Android software decoding/encoding`, most of times with videos that uses `c2.ffmpeg.vp9.decoder`, like this one: https://www.youtube.com/watch?v=6MeiED19LJk

KonstaT, 2024-12-16:
You would need to reboot your device for the video decoder option to take effect.

I tested with the above video and I can reproduce the issue with `c2.ffmpeg.vp9.decoder`. There's no issue when using `c2.android.vp9.decoder`. Also no issue with AV1 videos. So, some issue with FFmpeg's VP9 decoder. Maybe the length of the video (2+ hours in this case) has something to do with it as well.

You're welcome to dig into FFmpeg source code (https://github.com/raspberry-vanilla/android_external_ffmpeg/tree/android-15.0) or test if there's a previous version where this worked. Pull requests are welcome.

KonstaT, 2025-03-11:
Closing as there hasn't been any update in months. Most likely upstream FFmpeg issue.

Can be resolved by using Android's native VP9 decoder (`c2.android.vp9.decoder`).
https://github.com/raspberry-vanilla/android_local_manifest/wiki/Video-decoding-&-encoding