raspberry-vanilla/android_local_manifest #155: boot failed for the mounting of ext4 partition

tonyho, 2024-12-10:
I follow the README to build and generate(`rpi5-mkimg.sh`) the sdcard image, and tried to use dd and other GUI burn utils to burn the image to TF Card, but it fails to boot. 

Form the following detail log it seems the ext4 partition mounting is failed?

But I can mount all the partition (/dev/sdx1-4) on successfully on my Ubuntu Host, also tried use fsck/e2fsck to repair the partition, it still failed to boot? 

Could you kindly help to check the problem, thanks.

Both Android 14 and 15,  for both Rpi4b and rpi5b have problem.

Full log

```
  0.40 RPi: BOOTSYS release VERSION:6fe0b091 DATE: 2024/06/05 TIME: 16:41:49
  0.44 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1717602109 serial ac015ac7 boardrev d04170 stc 944566
  0.54 AON_RESET: 00000003 PM_RSTS 00001000
  0.62 RP1_BOOT chip ID: 0x20001927
  0.64 PM_RSTS: 0x00001000
  0.64 part 00000000 reset_info 00000000
  0.68 PMIC reset-event 00000000 rtc 00000000 alarm 00000000 enabled 0
  0.74 uSD voltage 3.3V
  1.94 Initialising SDRAM 'Samsung' 16Gb x2 total-size: 64 Gbit 4267
  1.97 DDR 4267 1 1 64 152
  2.42 OTP boardrev d04170 bootrom a a
  2.43 Customer key hash 0000000000000000000000000000000000000000000000000000000000000000
  2.50 VC-JTAG unlocked
  2.73 RP1_BOOT chip ID: 0x20001927
  4.28 SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
  4.39 SD HOST: 200000000 CTL0: 0x00800f00 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
  4.39 OCR c0ff8000 [300]
CID: 000353445344363447854ff0cb820186
CSD: 400e00325b590001dbd37f800a404000
  4.47 SD: bus-width: 4 spec: 2 SCR: 0x02458043 0x00000000
  4.57 SD HOST: 200000000 CTL0: 0x00800f04 BUS: 50000000 Hz actual: 50000000 HZ div: 4 (2) status: 0x1fff0000 delay: 2
  4.72 MBR: 0x00000800,  262144 type: 0x0c
  4.73 MBR: 0x00040800, 4194304 type: 0x83
  4.76 MBR: 0x00440800,  524288 type: 0x83
  4.80 MBR: 0x004c0800, 9697280 type: 0x83
  4.84 Trying partition: 0
  4.92 type: 32 lba: 2048 'mkfs.fat' ' boot       ' clusters 258078 (1)
  4.96 rsc 32 fat-sectors 2017 root dir cluster 2 sectors 0 entries 0
  4.02 FAT32 clusters 258078
  4.08 [sdcard] autoboot.txt not found
  4.08 Select partition rsts 0 C(boot_partition) 0 EEPROM config 0 result 0
  4.15 Trying partition: 0
  4.23 type: 32 lba: 2048 'mkfs.fat' ' boot       ' clusters 258078 (1)
  4.27 rsc 32 fat-sectors 2017 root dir cluster 2 sectors 0 entries 0
  4.33 FAT32 clusters 258078
  4.39 Read config.txt bytes      371 hnd 0xf4e8
  4.41 [sdcard] pieeprom.upd not found
  4.44 usb_max_current_enable default 0 max-current 3000
  4.61 Read bcm2712-rpi-5-b.dtb bytes    80149 hnd 0xf098
  4.63 dt-match: compatible: raspberrypi,5-model-b match: brcm,bcm2712
  4.69 dt-match: compatible: brcm,bcm2712 match: brcm,bcm2712
  4.76 Selecting USB low current limit

NOTICE:  BL31: v2.6(release):v2.6-239-g2a9ede0bd
NOTICE:  BL31: Built : 14:26:57, Jun 22 2023
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.6.56-gd6f8d77f6532-v8 (kleaf@build-host) (Android (11368308, +pgo, +bolt, +lto, +mlgo, based on r510928) clang version 18.0.0 (https://android.googlesourc4
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 5 Model B Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000002400000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x0000000002400000..0x00000000063fffff (65536 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x0000000000000000..0x000000000007ffff (512 KiB) nomap non-reusable atf@0
[    0.000000] OF: reserved mem: 0x000000003fd16260..0x000000003fd16297 (0 KiB) nomap non-reusable nvram@0
[    0.000000] Zone ranges:
[... 565 lines cut ...]
[    5.269794]  show_stack+0x28/0x3c
[    5.273114]  dump_stack_lvl+0x58/0x78
[    5.276785]  dump_stack+0x20/0x30
[    5.280106]  panic+0x148/0x35c
[    5.283165]  ext4_handle_error+0x1fc/0x224
[    5.287272]  __ext4_error+0xec/0x218
[    5.290854]  __ext4_new_inode+0xf24/0x112c
[    5.294960]  ext4_mkdir+0x13c/0x3d0
[    5.298455]  vfs_mkdir+0xf4/0x16c
[    5.301776]  do_mkdirat+0xa0/0x22c
[    5.305183]  __arm64_sys_mkdirat+0x4c/0x68
[    5.309290]  invoke_syscall+0x68/0x130
[    5.313047]  el0_svc_common+0xbc/0xf4
[    5.316717]  do_el0_svc+0x2c/0x3c
[    5.320038]  el0_svc+0x40/0x70
[    5.323097]  el0t_64_sync_handler+0x8c/0x108
[    5.327378]  el0t_64_sync+0x19c/0x1a0
[    5.331047] SMP: stopping secondary CPUs
[    5.334979] Kernel Offset: 0x1004000000 from 0xffffffc080000000
[    5.340916] PHYS_OFFSET: 0x0
[    5.343799] CPU features: 0x000002,00000001,70028143,0000720b
[    5.349562] Memory Limit: none
[    5.352621] Rebooting in 5 seconds..

```

KonstaT, 2024-12-12:
Have you tried using a different sdcard?

I'd also recommend against trying "repair" the file system as this can have the opposite effect.

tonyho, 2024-12-13:
> Have you tried using a different sdcard? Or properly formatting the sdcard including removing the partition table?
> 
> I'd also recommend against trying to "repair" the file system as this can have the opposite effect.

I do tried other sdcard, sandisk and others, but still the same results, could you kindly help to view the log and give some tips to clues?

I also tried use the balenaEtcher to burn which has the verification function.

wonschangge, 2024-12-16:
> It looks perfectly fine up to the point of
> 
> ```
> [    5.220666] EXT4-fs error (device mmcblk0p4): __ext4_new_inode:1283: comm init: failed to insert inode 12: doubly allocated?
> ```
> 
> Looks like you're the second person to come across this (https://forums.raspberrypi.com/viewtopic.php?t=323020&start=100#p2016138).
> 
> I didn't have any idea then, I don't have now either. I've suggested properly formatting the sdcard including removing the partition table (you can use erase sdcard option in Pi Imager or clear the partition table using GParted on Linux).

I missed the same question, "EXT4-fs error ... failed to insert inode 12: doubly allocated. ".
Your suggestion works fine, and I boot it successfully.

closed 2024-12-16 by tonyho