raspberry-vanilla/android_local_manifest #156: System date and time is not correct

demoa00, 2024-12-11:
The system take the image build date and time and not the current. I have investigate and i found that is not set any NTP server to retrieve the correct date and time. Is necessary to use the NTP service because the Raspberry has not a RTC module, integrated. I try some solutions but do not work. Do you have some ideas? Thanks.

KonstaT, 2024-12-12:
I can't reproduce any issue with network time. There's an option in the settings menu which is enabled by default.

I've heard some Chinese users have had issues with network time as Google's NTP servers are blocked by the great wall of China. See e.g. https://gist.github.com/xujiaao/63cb3bbea9fe22e79206e5eb7ba82d0e.

Pi 5 actually has RTC hardware and the kernel driver sets the system time accordingly.
```
[    1.733274] rpi-rtc soc:rpi_rtc: registered as rtc0
[    1.739675] rpi-rtc soc:rpi_rtc: setting system clock to 2024-12-12T12:03:15 UTC (1734004995)
```

demoa00, 2024-12-12:
Hi, i'm not from China but I reformulate my first question. What is the mechanism that Android Automotive use to set the date and time when there is any hardware components, such as an RTC, to use to set that informations? I suppose that in the absence of an hardware component, Android use the NTP service. Do you have any idea? Thanks.

KonstaT, 2024-12-12:
Yes, NTP server is used if Settings -> System -> Date & time -> Set time automatically is set. System time is set once network connection is established even if there's some RTC hardware available (i.e. you have wrong time stored on the RTC).

If there is no network connection or no RTC hardware, time is set by default to the build timestamp for the Android image. I'm not aware of this being any different Android Automotive.

KonstaT, 2025-01-08:
Closing as I can't reproduce any issue with network time. Assuming an issue with the network setup.

closed 2025-01-08 by KonstaT