raspberry-vanilla/android_local_manifest #158: Display doesn't seem to work on the latest version (r6) on RPI5

cburlacu, 2024-12-12:
Hi,

I built Android 15 (r6) for rpi5, but the display doesn't seem to work (hdmi or the official 7” DSI). If I revert this [commit](https://github.com/raspberry-vanilla/android_device_brcm_rpi5/commit/a47cb47fba5000e35de045915b55d5de3adf0d15), it works.

I've seen some `HWC3` errors in the logs, and by _diffing_ the generated vendor image with yours it looks like `HWC3`  service is not started ?

Thanks.

KonstaT, 2024-12-12:
I can't reproduce such issue on HDMI display. I don't have any DSI displays to test. You could also test my latest personal build (20241210) where I've switched to HWC3 (https://konstakang.com/devices/rpi5/AOSP15/). I haven't received any reports on display not working on that either.

Make sure your sources are in synced with the latest version and if the issue persists, please attach appropriate logs (`dmesg`, `logcat`)

cburlacu, 2024-12-12:
On your build it works ok after I modified the `config.txt` to configure it with the official display. Not sure yet what are the differences, but I can see these errors keeps repeating:
```
console:/ # [ 1371.878074] init: Control message: Could not find 'aidl/activity' for ctl.interface_start from pid: 206 (/system/bin/servicemanager)
[ 1372.458142] servicemanager: Caller(pid=609,uid=1000,sid=u:r:surfaceflinger:s0) Since 'android.hardware.graphics.composer3.IComposer/default' could not be found trying to start it as a lazy AIDL service. (if it's not configured to be a lazy service, it may be stuck starting or still starting).
[ 1372.484278] init: Control message: Could not ctl.interface_start for 'aidl/android.hardware.graphics.composer3.IComposer/default' from pid: 206 (/system/bin/servicemanager): File /vendor/bin/hw/android.hardware.composer.hwc3-service.drm(labeled "u:object_r:vendor_file:s0") has incorrect label or no domain transition from u:r:init:s0 to another SELinux domain defined. Have you configured your service correctly? https://source.android.com/security/selinux/device-policy#label_new_services_and_address_denials. Note: this error shows up even in permissive mode in order to make auditing denials possible.
[ 1372.537318] servicemanager: Caller(pid=609,uid=1000,sid=u:r:surfaceflinger:s0) Tried to start aidl service android.hardware.graphics.composer3.IComposer/default as a lazy service, but was unable to. Usually this happens when a service is not installed, but if the service is intended to be used as a lazy service, then it may be configured incorrectly.
[ 1372.747892] servicemanager: Caller(pid=8494,uid=1000,sid=u:r:system_server:s0) Since 'SurfaceFlingerAIDL' could not be found trying to start it as a lazy AIDL service. (if it's not configured to be a lazy service, it may be stuck starting or still starting).
```

closed 2024-12-12 by KonstaT

KonstaT, 2024-12-12:
Thanks for the report!

I had missed SELinux policy for the HWC3 service from https://github.com/raspberry-vanilla/android_external_drm_hwcomposer/blob/android-15.0/hwc3/hwc3-apex-file-contexts#L3. I've added it in the commits that closed the issue. Please sync your sources.

My personal builds have relaxed rules for services missing label which is why I didn't come across this.