raspberry-vanilla/android_local_manifest #159: tensorflow issue

aurosiva, 2024-12-13:
[image]
[image]
even after cloning tensorflow in external/ i get this issue while buiulding aosp 16 r6.

KonstaT, 2024-12-13:
Please use code tags and copy the terminal output as text. Can't copy&paste from screenshots.

I have `external/tensorflow` and `external/federated-compute` in my source tree and can't reproduce such issue.

If you're suspecting that you haven't synced the source code properly, use `repo sync`. Don't clone anything manually.

We're currently at AOSP 15.

aurosiva, 2024-12-14:
Hi @konsta

but i get these issue while syncing.

KonstaT, 2024-12-16:
> but i get these issue while syncing.

I'm not going to start guessing. But yeah, I wouldn't expect it to compile if you haven't even successfully synced the source code.

closed 2024-12-16 by KonstaT

aurosiva, 2024-12-16:
@Konsta ,

Thanks man