raspberry-vanilla/android_local_manifest #162: Can't build lunch aosp_rpi4_car-userdebug

vancodedekiru, 2024-12-18:
Hi,

I followed the instructions on this repository

https://github.com/raspberry-vanilla/android_local_manifest/tree/android-13.0

but every time, when I'm about 93% I encounter an error that stops the build.

Please see the error below:

ninja: build stopped: subcommand failed.
[  8% 1075/12199] //external/rust/crates/base64:libbase64_rust rustc src/lib.rs
[  8% 1076/12199] //external/rust/crates/base64:libbase64_rust rustc src/lib.rs [apex29]
ninja: build stopped: subcommand failed.
15:47:34 ninja failed with: exit status 1

 I am using Ubuntu 22.04.5 LTS laptop with 32gb of ram and 512gb of storage. 
 
 Did I miss something or are there dependencies that I need to install first?
 
 Thanks for your help.

KonstaT, 2024-12-18:
As noted, Raspberry Vanilla `android-13.0` branch is no longer maintained. Consider using newer AOSP versions.

The actual error message is in the output somewhere before what you copy&pasted here.

vancodedekiru, 2024-12-19:
Hi,

I tried to build from android 14 branch

https://github.com/raspberry-vanilla/android_local_manifest/tree/android-14.0

but it still failed at 93%

here is the error further back from what I pasted before

[ 93% 132487/142034] target Prebuilt: libidmap2_policies (out/target/product/rpi4/obj/SHARED_LIBRARIES/libidmap2_policies_intermediates/libidmap2_policies.so)
[ 93% 132488/142034] //frameworks/base/cmds/idmap2:libidmap2_policies generate toc libidmap2_policies.so
[ 93% 132489/142034] //external/rust/crates/libz-sys:liblibz_sys rustc src/lib.rs
[ 93% 132490/142034] Copy symbols with mapping: out/target/product/rpi4/symbols/system/lib64/liblibz_sys.dylib.so
[ 93% 132491/142034] build out/target/product/rpi4/obj/MESON_LIBCAMERA/.build.timestamp
FAILED: out/target/product/rpi4/obj/MESON_LIBCAMERA/.build.timestamp
/bin/bash -c "(rm -rf out/target/product/rpi4/obj/MESON_LIBCAMERA/ ) && (mkdir -p out/target/product/rpi4/obj/MESON_LIBCAMERA/ ) && (mkdir -p out/target/product/rpi4/obj/MESON_LIBCAMERA//build ) && (cp -r external/libcamera//* out/target/product/rpi4/obj/MESON_LIBCAMERA/ ) && (cd out/target/product/rpi4/obj/MESON_LIBCAMERA && PATH=/usr/bin:/usr/local/bin:\$PATH 
.........
AttributeError: module 'jinja2' has no attribute 'Environment'
[10/167] Compiling C++ object src/libcamera/base/libcamera-base.so.p/class.cpp.o
[11/167] Generating src/ipa-priv-key with a custom command
.....+...+........+..........+............+........+...+....+.....+......+..........+.....+.......+.....+.......+............+...+..+.+........+......+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++*.+......+..............+............+.........+.+...+.........+.....+.......+.........+.........+......+......+...+...........+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++*...+...+.....+............+.......+........+.......+......+..+.......+.....+..........+.....+.......+........+.......+...........+.......+..........................+......+.+.....+....+...+.......................+...+.+......+........+......+.............+..+..........+.........+..+...+......+.+..+...+.........+...+.+..+.............+......+...+............+.....+...+...+....+...+...+..+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
.+...........+...+.......+...+.....+......+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++*..+......+........+..................+....+...+...+...+..+.......+.....+...+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++*..........+.....+.+............+..+...+.......+.....+.......+........+.+.....+.......+..+..........+..+......+.......+...............+......+..+...............+.........+...............+....+......+.....+...+.............+...+.........+..+....+...........+................+...+.....+....+...+........+.........+.+..+...+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
[12/167] Generating include/libcamera/control_ids_h with a custom command
[13/167] Compiling C++ object src/libcamera/base/libcamera-base.so.p/semaphore.cpp.o
[14/167] Compiling C++ object src/libcamera/base/libcamera-base.so.p/event_dispatcher.cpp.o
[15/167] Compiling C++ object src/libcamera/base/libcamera-base.so.p/message.cpp.o
[16/167] Compiling C++ object src/libcamera/base/libcamera-base.so.p/bound_method.cpp.o
[17/167] Compiling C++ object src/libcamera/base/libcamera-base.so.p/event_notifier.cpp.o
[18/167] Compiling C++ object src/libcamera/base/libcamera-base.so.p/shared_fd.cpp.o
[19/167] Compiling C++ object src/libcamera/base/libcamera-base.so.p/backtrace.cpp.o
[20/167] Compiling C++ object src/libcamera/base/libcamera-base.so.p/file.cpp.o
[21/167] Compiling C++ object src/libcamera/base/libcamera-base.so.p/event_dispatcher_poll.cpp.o
[22/167] Compiling C++ object src/libcamera/base/libcamera-base.so.p/log.cpp.o
ninja: build stopped: subcommand failed.
[ 93% 132492/142034] build out/target/product/rpi4/obj_arm/MESON_LIBCAMERA/.build.timestamp
FAILED: out/target/product/rpi4/obj_arm/MESON_LIBCAMERA/.build.timestamp
/bin/bash -c "(rm -rf out/target/product/rpi4/obj_arm/MESON_LIBCAMERA/ ) && (mkdir -p out/target/product/rpi4/obj_arm/MESON_LIBCAMERA/ ) && (mkdir -p out/target/product/rpi4/obj_arm/MESON_LIBCAMERA//build ) && (cp -r external/libcamera//* out/target/product/rpi4/obj_arm/MESON_LIBCAMERA/ ) && (cd out/target/product/rpi4/obj_arm/MESON_LIBCAMERA &&

KonstaT, 2024-12-19:
Have you followed the instructions on how to setup the build environment?

Step 2. Install additional packages:
```
sudo pip3 install meson mako jinja2 ply pyyaml dataclasses
```

losgobbi, 2024-12-20:
Just for the record, I'm also trying to build, but android-14.0.0_r22 branch/aosp_rpi4-userdebug product. Despite that I've faced the same error regarding jinja2, now, some external projects doesn't compile:

`error: external/drm_hwcomposer/Android.bp:271:1: "com.android.hardware.graphics.composer.drm_hwcomposer" depends on undefined module "com.android.hardware.key".
`
Even bypassing this, forcing a specific sha at hwcomposer, the build process continues but it fails at another point (mesa in this case):
`meson.build:4:0: ERROR: Meson version is 0.61.2 but project requires >= 1.1.0
`
It looks like that some repos have moved forward/fetching new things but are not compatible anymore with the rest of the AOSP tree. Is this true or I've missed something here? 
Tks and regards.

KonstaT, 2024-12-20:
> I'm also trying to build, but android-14.0.0_r22 

Why `android-14.0.0_r22`? Using outdated release tags is not supported nor recommended in any way. I've added a note to the build documentation to reflect this.

> Despite that I've faced the same error regarding jinja2

Maybe it would benefit this specific issue if you mentioned how you resolved this? Assuming followed the instructions on how to setup the build environment as above?

> some external projects doesn't compile

Duplicate of https://github.com/raspberry-vanilla/android_local_manifest/issues/160 & https://github.com/raspberry-vanilla/android_local_manifest/issues/161.

> the build process continues but it fails at another point (mesa in this case):

The error message `Meson version is 0.61.2 but project requires >= 1.1.0` really couldn't be more self-explanatory. Please update your `meson`.

> It looks like that some external repos have moved forward/fetching new things but are not compatible anymore with the rest of the AOSP tree. Is this true or I've missed something here?

As mentioned elsewhere, you can see from the device manifest (https://github.com/raspberry-vanilla/android_local_manifest/blob/android-14.0.0_r22/manifest_brcm_rpi.xml) what repositories are frozen to `android-14.0.0_r22` and what are receiving further updates (i.e. `android-14.0`). The correct solution would be to just use the `android-14.0` branch instead.

losgobbi, 2024-12-20:
> Why android-14.0.0_r22
It was an arbitrary test over A14 and I didn't known that r22 branch was not maintained/valid anymore. I was just looking for an A14 baseline.

> Assuming followed the instructions on how to setup the build environment as above?
Yes, I just did the pip install command.

> Please update your meson.
Yes, I understood the error message, but the problem is what is the possible commit point that is valid to build the mesa inside the tree. Actually, I did a couple of tests without positive results.

>  The correct solution would be to just use the android-14.0 branch instead.
Yeah, I'll try that and see how it goes.

Tks for the answer and regards!!!

KonstaT, 2024-12-20:
> Yes, I understood the error message, but the problem is what is the possible commit point that is valid to build the mesa inside the tree. Actually, I did a couple of tests without positive results.

Sorry, don't quite understand the question. Mesa 24.1 and up require `meson` >= 1.1.0 (https://github.com/raspberry-vanilla/android_external_mesa3d/blob/android-14.0-24.1/meson.build#L26 vs. https://github.com/raspberry-vanilla/android_external_mesa3d/blob/android-14.0-24.0/meson.build#L26). This is about the build tools on your host machine and not so much about what Mesa version was used at what point.

Building something with `meson` has never been part of AOSP. It's a HACK that Mesa project adopted since it became too tedious the maintain proper Android makefiles. Google used NDK to build Mesa for devices that utilized it and now with Android 15 QPR1 they've moved to using meson2hermetic to generate proper blueprint makefiles for the parts that they need (https://android.googlesource.com/platform/external/mesa3d/+/refs/tags/android-15.0.0_r6/meson_to_hermetic/, https://github.com/raspberry-vanilla/android_external_mesa3d/commit/a6baafe22db3ad9ecec237796f75818e900e3692).

losgobbi, 2024-12-23:
> This is about the build tools on your host machine and not so much about what Mesa version was used at what point.

Yes, I made a confusion regarding that. I was trying to make the external/mesa subproject compatible with my host meson version rather the opposite. I'll try to check that too.
Tks for pointing that. Regards.

KonstaT, 2025-01-08:
Closing as there has been no update in nearly a month. Assuming resolved by installing `jinja2` is instructed in the documentation.

closed 2025-01-08 by KonstaT