raspberry-vanilla/android_local_manifest #163: Build android 12 automotive

Sotatek-ChienNguyen2, 2024-12-24:
Hi friend,
I want build android 12 automotive

RealaxER, 2024-12-27:
Hi if you want build aosp 12 for automotive, you can try this link: https://grapeup.com/blog/android-automotive-os-on-raspberry-pi-4b/
But if you want to use this source you will have to add launcher and some automotive services
(Nếu mới bắt đầu thì dùng src trên)

Sotatek-ChienNguyen2, 2024-12-27:
Theo guide này mình thấy nó là android 11. Liệu có phải là muốn build android 12 automotive thì sẽ build từ android aosp 12 cho PI4 rồi port thêm một số dịch vụ của bản automotive vào phải không ạ

RealaxER, 2024-12-27:
Tự port không phải dễ, b có thể try source 11 trước, 2 bọn src cũng chả khác nhau gì mấy 
Tại sao phải là 12 ?

KonstaT, 2024-12-30:
Android Automotive build target is supported on Raspberry Vanilla project on AOSP13 and forward.

There really is no reason to use several years old Android version at this point.

closed 2024-12-30 by KonstaT