raspberry-vanilla/android_local_manifest #164: Reboot to recovery not working (RPi 5)

Nickldio2, 2025-01-01:
Hi, first of all thank you for creating this build, I am very grateful for that.

I am trying to boot the device into TWRP recovery mode, but can't seem to work it out. I have compiled a ramdisk-recovery image according to your instructions (https://github.com/twrp-raspberry/android_local_manifest) and added that to the boot partition.

I have tried doing a recovery reboot via adb shell and also powering the device with GPIO 21 connected to ground but both do not seem to work. In your version of AOSP 14 you have added an option which enables the boot to recovery in the android UI. Could you share how you achieved that or alternatively how I might enter TWRP recovery?*

Cheers

KonstaT, 2025-01-02:
It's not in scope of Raspberry Vanilla project to support Android recovery mode. E.g. https://github.com/twrp-raspberry/android_local_manifest/issues would be a better suited place for this discussion.

AOSP doesn't actually have any reboot to recovery option. Following is executed when you reboot to recovery on my personal builds.
mv /boot/ramdisk.img /boot/ramdisk-boot.img
mv /boot/ramdisk-recovery.img /boot/ramdisk.img
sed -i "s/^dtoverlay=vc4-kms-v3d$/#dtoverlay=vc4-kms-v3d/" /boot/config.txt

Booting out of TWRP does the opposite.
https://github.com/twrp-raspberry/android_device_brcm_rpi/blob/twrp-11/rebootrpi.sh#L8-L10

You can also check /boot/config.txt on my personal builds to see what the GPIO21 logic does.

closed 2025-01-02 by KonstaT

Nickldio2, 2025-01-02:
Okay, thank you very much, will look into it!