raspberry-vanilla/android_local_manifest #169: rpi5 android 15 aosp_rpi5_car build issue

sharaf-ali, 2025-01-06:
Hello,

I am facing a build issue for rpi5 android 15 aosp_rpi5_car 

getting following fail message 

[100% 131/131] analyzing Android.bp files and generating ninja file at out/soong/build.aosp_rpi5_car.ninja
FAILED: out/soong/build.aosp_rpi5_car.ninja
cd "$(dirname "out/host/linux-x86/bin/soong_build")" && BUILDER="$PWD/$(basename "out/host/linux-x86/bin/soong_build")" && cd / && env -i  "$BUILDER"     --top "$TOP"     --soong_out "out/soong"     --out
 "out"     --soong_variables out/soong/soong.aosp_rpi5_car.variables -o out/soong/build.aosp_rpi5_car.ninja -l out/.module_paths/Android.bp.list --available_env out/soong/soong.environment.available --use
d_env out/soong/soong.environment.used.aosp_rpi5_car.build Android.bp
Warning: Module 'androidx.wear_wear' depends on non-existing optional_uses_libs 'wear-sdk'
Warning: Module 'androidx.wear.compose_compose-foundation' depends on non-existing optional_uses_libs 'wear-sdk'
Killed
22:20:37 soong bootstrap failed with: exit status 1
ninja: build stopped: subcommand failed.

#### failed to build some targets (03:10 (mm:ss)) ####

if anybody have solution for this please let me know.

Regards
Sharaf ali

KonstaT, 2025-01-07:
Looks very similar to https://github.com/raspberry-vanilla/android_local_manifest/issues/128. I assumed this failed on out of memory.

Actual error message could be somewhere before what you copy&pasted here or you would need to check soong logs.

KonstaT, 2025-01-31:
Closing as there hasn't been any update in a month and no further information was provided. Assuming resolved using a build machine that meets the hardware requirements.

closed 2025-01-31 by KonstaT