raspberry-vanilla/android_local_manifest #172: failed build Interface android.media.audio.common.types can not be marked frozen: false

purnomoshinigami, 2025-01-14:
we need help raspi5 
if we use  android-15.0.0_r6 is working 
but if we use android-15.0.0_r10 to 13  
we have error  compiling  at 3%  

FAILED: out/soong/.intermediates/system/hardware/interfaces/media/android.media.audio.common.types-api/has_development
rm -f out/soong/.intermediates/system/hardware/interfaces/media/android.media.audio.common.types-api/has_development && out/host/linux-x8
6/bin/aidl --checkapi=equal --stability vintf system/hardware/interfaces/media/aidl_api/android.media.audio.common.types/3 out/soong/.int
ermediates/system/hardware/interfaces/media/android.media.audio.common.types-api/dump 2> /dev/null && echo "Interface android.media.audio
.common.types can not be marked \frozen: false\ if there are no changes between the current version and the last frozen version." && ex
it -1 || echo $? > out/soong/.intermediates/system/hardware/interfaces/media/android.media.audio.common.types-api/has_development # hash 
of input list: e38daf1b4abf912a9f79077d376ed26dfa23cdb59efb4178d0dfb56ed910a3e9
Interface android.media.audio.common.types can not be marked frozen: false if there are no changes between the current version and the 
last frozen version.
23:05:45 ninja failed with: exit status 1
There was 1 action that completed after the action that failed. See verbose.log.gz for its output.

KonstaT, 2025-01-15:
Please follow the instructions on how to initialize repo. There's exactly one AOSP release tag that is supported at a time. Currently it's `android-15.0.0_r10`.

I can't reproduce such issue on `android-15.0.0_r10`. Differences upstream to last month's `android-15.0.0_r6` are also minimal so I can't see why that would work any better. More likely some issue with e.g. syncing your sources.

purnomoshinigami, 2025-01-15:
I followed the instructions when the repo was updated to android-15.0.0_r10, 

we got that error, so we tried changing the base repo to 15.0.0_r11, 15.0.0_r12, 15.0.0_r13
 to find out where the error was, do I need to update for that? , we think it might be a bug of aosp .
 and we always try to compile it the first time.
 After that we make the modifications we need and we get an error the first time after syncing.
 We will look for the problem first, if later we find the problem where we will update it. 
 
Thank you for your attention.

KonstaT, 2025-01-16:
If it was an AOSP issue, I would be able to reproduce it. I can't, so more likely an issue at your end.

If I need guess, either you've ignored some errors during `repo sync` or you're attempting to build something other what is described in the project documentation (`make bootimage systemimage vendorimage -j$(nproc)`).

purnomoshinigami, 2025-01-16:
The issue is with my team. aosp folder should symslink to android-15.0.0_r10. 
and sync the repo to the aosp folder.
but my team created a folder android-15.0.0_r10 and synced into it, and the folder was renamed to aosp. Once again we apologize.

closed 2025-01-16 by purnomoshinigami