raspberry-vanilla/android_local_manifest #173: android-14.0 branch build failed

hkrishna-git, 2025-01-15:
- **Environment** :
   
  Ubuntu 24.04.1 LTS (GNU/Linux 5.15.167.4-microsoft-standard-WSL2 x86_64)

All the necessary tools mentioned in the steps are succeeded.
repo init & sync also succeeded

- **Build configuration / commands used** :
   lunch aosp_rpi4_car-ap2a-userdebug
   make bootimage systemimage vendorimage -j$(nproc)

Attached build failure log file for reference.
Could you please let me know how to fix this ?

- **Below build error observed in the attached log** :
  
   Code processing error in frameworks/base/libs/WindowManager/Shell/src/com/android/wm/shell/bubbles/BubbleStackView.java:453:
   Java parsing error: (line 453,col 72) Parse error. Found  "view" <IDENTIFIER>, expected one of  "!=" "%=" "&" "&&" "&=" ")" "*=" "+=" "-=" "->" "/=" "::" "<<=" "=" "==" ">>=" ">>>=" "?" "^" "^=" "|" "|=" "||"

- **Build Log** :
   
     [build.log](https://github.com/user-attachments/files/18419391/build.log)

hkrishna-git, 2025-01-15:
- **Java version installed** :
     java --version
    openjdk 21.0.5 2024-10-15
    OpenJDK Runtime Environment (build 21.0.5+11-Ubuntu-1ubuntu124.04)
    OpenJDK 64-Bit Server VM (build 21.0.5+11-Ubuntu-1ubuntu124.04, mixed mode, sharing)

KonstaT, 2025-01-15:
I'm not a Windows user and I have no idea if you're even supposed to be able to build AOSP using WSL2.

Recommended native build environment (Ubuntu 22.04 LTS) is mentioned in the project's documentation.
https://github.com/raspberry-vanilla/android_local_manifest

hkrishna-git, 2025-01-16:
Thanks for the prompt reply.

Yes, I see it is recommended to use native Ubuntu 22.04 but with the same environment with branch "android-13.0" build was successful. As I see few comments states that "android-13.0" branch is not supported for the queries so I thought to use "android-14.0" branch.

Does anyone have the same issue ? Could you please let me know how to fix this ?

hkrishna-git, 2025-01-21:
We can close this issue as it was able to build as mentioned in the above comment

closed 2025-01-21 by hkrishna-git