raspberry-vanilla/android_local_manifest #174: rpi4 android-15.0.0_r10 android.hardware.usb.gadget-service.rpi crash --> device not booting

DerTeufel, 2025-01-17:
Hi,

initially i tried building/booting aosp_rpi4_car-ap4a, but as this (also) didn't boot, this log now is from aosp_rpi4-ap4a-eng build:

`01-01 00:00:51.320   564   564 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
01-01 00:00:51.320   564   564 F DEBUG   : Build fingerprint: 'Raspberry/aosp_rpi4/rpi4:15/AP4A.250105.002/eng.www-da:eng/test-keys'
01-01 00:00:51.320   564   564 F DEBUG   : Revision: '0'
01-01 00:00:51.320   564   564 F DEBUG   : ABI: 'arm64'
01-01 00:00:51.320   564   564 F DEBUG   : Timestamp: 1970-01-01 00:00:51.067867025+0000
01-01 00:00:51.320   564   564 F DEBUG   : Process uptime: 3s
01-01 00:00:51.320   564   564 F DEBUG   : Cmdline: /vendor/bin/hw/android.hardware.usb.gadget-service.rpi
01-01 00:00:51.320   564   564 F DEBUG   : pid: 361, tid: 361, name: android.hardwar  >>> /vendor/bin/hw/android.hardware.usb.gadget-service.rpi <<<
01-01 00:00:51.320   564   564 F DEBUG   : uid: 1000
01-01 00:00:51.320   564   564 F DEBUG   : tagged_addr_ctrl: 0000000000000001 (PR_TAGGED_ADDR_ENABLE)
01-01 00:00:51.320   564   564 F DEBUG   : signal 6 (SIGABRT), code -1 (SI_QUEUE), fault addr --------
01-01 00:00:51.320   564   564 F DEBUG   :     x0  0000000000000000  x1  0000000000000169  x2  0000000000000006  x3  0000007fe280a0c0
01-01 00:00:51.321   564   564 F DEBUG   :     x4  0000000000000010  x5  0000000000000010  x6  0000000000000010  x7  7f7f7f7f7f7f7f7f
01-01 00:00:51.321   564   564 F DEBUG   :     x8  00000000000000f0  x9  1a807c78e2a06a5b  x10 000000ff00000020  x11 fffffffffffffffd
01-01 00:00:51.321   564   564 F DEBUG   :     x12 0000007fe2809bb0  x13 000000000000001c  x14 0000007fe2809d18  x15 00000000a3f25fa0
01-01 00:00:51.321   564   564 F DEBUG   :     x16 0000007e18d8b048  x17 0000007e18d74240  x18 0000007e1f770000  x19 0000000000000169
01-01 00:00:51.321   564   564 F DEBUG   :     x20 0000000000000169  x21 00000000ffffffff  x22 0000007fe280a328  x23 0000007fe280a330
01-01 00:00:51.321   564   564 F DEBUG   :     x24 0000000000000000  x25 0000000000000000  x26 0000000000000000  x27 0000000000000000
01-01 00:00:51.321   564   564 F DEBUG   :     x28 0000000000000000  x29 0000007fe280a140
01-01 00:00:51.321   564   564 F DEBUG   :     lr  0000007e18d13230  sp  0000007fe280a0c0  pc  0000007e18d13254  pst 0000000000001000
01-01 00:00:51.321   564   564 F DEBUG   : 5 total frames
01-01 00:00:51.321   564   564 F DEBUG   : backtrace:
01-01 00:00:51.321   564   564 F DEBUG   :       #00 pc 000000000005b254  /apex/com.android.runtime/lib64/bionic/libc.so (abort+156) (BuildId: 683ee52472ca2c971bee317bbfcf2db3)
01-01 00:00:51.322   564   564 F DEBUG   :       #01 pc 000000000000500c  /vendor/bin/hw/android.hardware.usb.gadget-service.rpi (aidl::android::hardware::usb::gadget::UsbGadget::UsbGadget()+100) (BuildId: 66ee4ea7d1e90a53525e0eed78e0320e)
01-01 00:00:51.322   564   564 F DEBUG   :       #02 pc 00000000000049a4  /vendor/bin/hw/android.hardware.usb.gadget-service.rpi (std::__1::shared_ptr<aidl::android::hardware::usb::gadget::UsbGadget> ndk::SharedRefBase::make<aidl::android::hardware::usb::gadget::UsbGadget>()+48) (BuildId: 66ee4ea7d1e90a53525e0eed78e0320e)
01-01 00:00:51.322   564   564 F DEBUG   :       #03 pc 00000000000047cc  /vendor/bin/hw/android.hardware.usb.gadget-service.rpi (main+48) (BuildId: 66ee4ea7d1e90a53525e0eed78e0320e)
01-01 00:00:51.322   564   564 F DEBUG   :       #04 pc 0000000000054e98  /apex/com.android.runtime/lib64/bionic/libc.so (__libc_init+116) (BuildId: 683ee52472ca2c971bee317bbfcf2db3)`

there seem to be more issues later down the log, hence i pasted the whole log here:
https://gist.github.com/DerTeufel/bd4c7af31112321237875f0d4c2aa41d

let me know, if you need anything else

KonstaT, 2025-01-17:
IIRC android.hardware.usb.gadget-service.rpi crashes because the service starts before `sys.usb.controller` property is set and functionfs is mounted (https://github.com/raspberry-vanilla/android_device_brcm_rpi4/blob/android-15.0/ramdisk/init.rpi4.usb.rc#L85-L89). It doesn't prevent the device from and booting or the USB gadget from working. The service restarts until it's successful.

Actually, I don't see anything that would suggest that device doesn't boot successfully eventually. I can see that you don't have any display connected (or at least one that reports resolutions it supports via EDID). Maybe a blank display is what causes you to assume it doesn't boot?

DerTeufel, 2025-01-17:
> Maybe a blank display is what causes you to assume it doesn't boot?

oh... this is awkward... i am using the original raspberry 7" touch display, connected via that ribbon cable.
This isn't supported? or do i need to change something?

KonstaT, 2025-01-17:
https://github.com/raspberry-vanilla/android_local_manifest/wiki/DSI-display

DerTeufel, 2025-01-17:
thank you so much.
it's been too long that i didn't work with my pi. I'll rebuild and confirm it works as expected, then close the issue

closed 2025-01-17 by DerTeufel