raspberry-vanilla/android_local_manifest #176: Large system size >2GB is not possible

Frank10021960, 2025-01-22:
Create a system image larger than 2GB should be possible. For example for some large vendor Apps.
BOARD_SYSTEMIMAGE_PARTITION_SIZE is normally the right variable, but > 2GB imga edoes not start correct.

KonstaT, 2025-01-22:
Duplicate of https://github.com/raspberry-vanilla/android_local_manifest/issues/137.

https://github.com/raspberry-vanilla/android_device_brcm_rpi5/blob/android-15.0/BoardConfig.mk#L58 & https://github.com/raspberry-vanilla/android_device_brcm_rpi5/blob/android-15.0/mkimg.sh#L31.

closed 2025-01-22 by KonstaT

Frank10021960, 2025-01-22:
I cannot see, that the problem is fixed. Max is currently only 2GB. When you adjust a higher value, the system goes in a restart loop again and again.

KonstaT, 2025-01-22:
Yet setting a larger a system partition size works just fine.

What did you modify exactly attempting to "fix" this?

Please attach appropriate logs (dmesg, logcat). Are you sure your bootloop isn't because of some other modifications you've made (that required the larger system partition size in the first place)?

Frank10021960, 2025-01-22:
I have only changed the value from
# old FRadtke BOARD_SYSTEMIMAGE_PARTITION_SIZE := 2147483648 # 2048M
to
BOARD_SYSTEMIMAGE_PARTITION_SIZE := 4294967296
in BoardConfig.mk, so that it should possible to have a large image.
To get a large image, I copy a large file to product in the build
#FRadtke test large image
PRODUCT_COPY_FILES += \
    device/brcm/rpi5/test/system_test.img:product/media/system_test.img
in aosp_rpi5_car.mk

KonstaT, 2025-01-22:
https://github.com/raspberry-vanilla/android_local_manifest/issues/176#issuecomment-2607008296