raspberry-vanilla/android_local_manifest #179: Building failed on libcamera.so

promfin, 2025-01-31:
Hello

============================================
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=14
PRODUCT_INCLUDE_TAGS=com.android.mainline mainline_module_prebuilt_nightly
TARGET_PRODUCT=aosp_rpi5
TARGET_BUILD_VARIANT=userdebug
TARGET_ARCH=arm64
TARGET_ARCH_VARIANT=armv8-a
TARGET_CPU_VARIANT=cortex-a76
TARGET_2ND_ARCH=arm
TARGET_2ND_ARCH_VARIANT=armv7-a-neon
TARGET_2ND_CPU_VARIANT=generic
HOST_OS=linux
HOST_OS_EXTRA=Linux-5.15.0-131-generic-x86_64-Ubuntu-22.04.5-LTS
HOST_CROSS_OS=windows
BUILD_ID=AP2A.240905.003
OUT_DIR=/home/out

After 8 minutes of build received this error:

FAILED: ninja: 'home/out/target/product/rpi5/obj_arm/MESON_LIBCAMERA/install/usr/local/lib/libcamera-hal.so', needed by '/home/out/target/product/rpi5/obj_arm/SHARED_LIBRARIES/camera.libcamera_intermediates/LINKED/camera.libcamera.so', missing and no known rule to make it
09:47:12 ninja failed with: exit status 1

#### failed to build some targets (08:09 (mm:ss)) ####

Tried few times on different servers. What can be wrong?

KonstaT, 2025-01-31:
Possibly a duplicate of https://github.com/raspberry-vanilla/android_local_manifest/issues/74.

Have you set `OUT_DIR` environment variable? `OUT_DIR=/home/out` seems odd as `OUT_DIR=out` is expected. Mesa and libcamera are built with external meson build system which doesn't support using custom output directory.

Also the actual the error message is somewhere before what you copy&pasted here.

KonstaT, 2025-02-12:
Closing as there's been no update in a couple of weeks. Assuming most likely resolved by not setting custom `OUT_DIR`.

closed 2025-02-12 by KonstaT