raspberry-vanilla/android_local_manifest #182: [RPI4B] CarSettings App is not visible in Android Apps by default

hkrishna-git, 2025-02-07:
Hi Team,

Observed that "CarSettings" app is not installed by default.
Used the branch : android-14.0 (android-14.0.0_r67).

Though it is added in "PRODUCT_PACKAGES" (packages/services/Car/car_product/build/car.mk)

But listed in the packages list :

rpi4:/ # pm list packages | grep car.settings
package:com.android.car.settings

Is anyone facing the same issue ?
Please let me know how to resolve it.

Thanks in advance.

KonstaT, 2025-02-07:
> Hi Team

Hi. It's just me. There is no team in I.

> Observed that "CarSettings" app is not installed by default.

Observed how?

> ...

Doesn't this exactly indicate that it is installed. I'm also sure there's a settings menu on Android Automotive, so...

Maybe you're looking for Settings -> Apps -> three dot menu -> Show system (or whatever the equivalent is on Android Automotive).

In any case, closing as this is not an issue with the Raspberry Pi device specific implementation that this project provides.

closed 2025-02-07 by KonstaT

hkrishna-git, 2025-02-13:
Hi,

Sorry for the late reply on this issue.

Actually I mean, "CarSettings" app lists in the packages list but doesn't visible in the application list in HMI
For more information, please find the attached screenshot of the RPI HMI Screen.

[image]

Could you please let check once ?

Thank you.

KonstaT, 2025-02-13:
Scroll. S is in alphabets after L where this screenshot ends.

hkrishna-git, 2025-02-13:
hmm there is no scrolling available in the HMI