raspberry-vanilla/android_local_manifest #183: Android 15 crash after increasing system partition size

aladin153, 2025-02-20:
After adding some priv-app apk to the android 15 (automotive) system partition, the default 2048MB  size is no more sufficient to hold all contents.
After increasing the system partition size to 2560Mb (in BoardConfig.mk), a lot of crashes occurs during bootup.

KonstaT, 2025-02-20:
Duplicate of https://github.com/raspberry-vanilla/android_local_manifest/issues/176 & https://github.com/raspberry-vanilla/android_local_manifest/issues/137.

closed 2025-02-20 by KonstaT

aladin153, 2025-02-21:
Even after increasing the system partition size in both files BoardConfig.mk and mkimg.sh, zygote is always crashing.
Please find attached the dmesg and logcat.

[dmesg.txt](https://github.com/user-attachments/files/18903856/dmesg.txt)

[logcat.txt](https://github.com/user-attachments/files/18904039/logcat.txt)

KonstaT, 2025-02-21:
Did you compile both system.img and the flashable image after changing the size?

Are you sure this is not related to "adding some priv-app apk"? I have tested increasing the system partition size earlier this week so I'm absolutely sure it works.

aladin153, 2025-02-21:
Yes, i compiled all images (vendor, system and boot).
Even with a clean build (without any app added), increasing system size causes crash.

Target: aosp_rpi5_car-ap4a-userdebug
HW: RPI5 8Gb

KonstaT, 2025-02-22:
Did you do build without any modification and test that it booted? What exactly did you change?

2560M = 2560*1024*1024 = 2684354560 bytes.

Still, I'm absolutely sure there's no issue increasing the system partition size as I just did this myself this week.

aladin153, 2025-02-24:
Hello,

The build works perfectly now after increasing the system partition size.

Raspberry vanilla build (without any modification) : OK
Raspberry vanilla build (+ just increase system partition size to 2560M) : OK

After that i start applying my customization's one by one, and finally i obtain a working build.
I'm trying to understand the crash root cause, because i just replaced an Android.mk with Android.bp and removed an xml permission file.

In the same subject, is it right to keep the userdata_partition_size at 128M even if the physical emmc memory for this partition is up to 3 Gb ?

Thanks,
Alaeddine ZAYEN.