raspberry-vanilla/android_local_manifest #185: Android 15 Camera Issue rpi5

danascape, 2025-02-27:
I am running both the official v2 and v3 cameras. They both typically fail to capture a snapshot a second time in the same capture session. However, the v3 camera will capture a full resolution snapshot, whereas the v2 camera will only capture up to 1920x1080.

Same happens with arducam

With the CSI cameras, live view is working and no snapshot is in progress.

2025-02-10 19:10:54.782   383-397   libcamera               and...amera.provider@2.5-service_64  E  ERROR Buffer mapped_framebuffer.cpp:230 Failed to mmap plane: Permission denied
2025-02-10 19:10:54.884   383-397   libcamera               and...amera.provider@2.5-service_64  E  ERROR Buffer mapped_framebuffer.cpp:230 Failed to mmap plane: Permission denied
2025-02-10 19:10:55.179   383-397   libcamera               and...amera.provider@2.5-service_64  E  ERROR Buffer mapped_framebuffer.cpp:230 Failed to mmap plane: Permission denied
2025-02-10 19:10:55.680   383-397   libcamera               and...amera.provider@2.5-service_64  E  ERROR Buffer mapped_framebuffer.cpp:230 Failed to mmap plane: Permission denied
2025-02-10 19:10:55.782   383-397   libcamera               and...amera.provider@2.5-service_64  E  ERROR Buffer mapped_framebuffer.cpp:230 Failed to mmap plane: Permission denied
2025-02-10 19:10:55.884   383-397   libcamera               and...amera.provider@2.5-service_64  E  ERROR Buffer mapped_framebuffer.cpp:230 Failed to mmap plane: Permission denied
2025-02-10 19:10:56.384   383-397   libcamera               and...amera.provider@2.5-service_64  E  ERROR Buffer mapped_framebuffer.cpp:230 Failed to mmap plane: Permission denied
2025-02-10 19:10:56.584   383-397   libcamera               and...amera.provider@2.5-service_64  E  ERROR Buffer mapped_framebuffer.cpp:230 Failed to mmap plane: Permission denied
2025-02-10 19:10:56.776   721-1925  BufferPoolAccessor2.0   media.swcodec                        D  bufferpool2 0xb400007c463dd538 : 58(826016 size) total buffers - 1(1971 size) used buffers - 495/1799 (recycle/alloc) - 1321/1798 (fetch/transfer)
2025-02-10 19:10:56.782   383-397   libcamera               and...amera.provider@2.5-service_64  E  ERROR Buffer mapped_framebuffer.cpp:230 Failed to mmap plane: Permission denied
2025-02-10 19:10:56.884   383-397   libcamera               and...amera.provider@2.5-service_64  E  ERROR Buffer mapped_framebuffer.cpp:230 Failed to mmap plane: Permission denied
2025-02-10 19:10:56.981   383-397   libcamera               and...amera.provider@2.5-service_64  E  ERROR Buffer mapped_framebuffer.cpp:230 Failed to mmap plane: Permission denied
2025-02-10 19:10:57.182   383-397   libcamera               and...amera.provider@2.5-service_64  E  ERROR Buffer mapped_framebuffer.cpp:230 Failed to mmap plane: Permission denied
2025-02-10 19:10:57.479   383-397   libcamera               and...amera.provider@2.5-service_64  E  ERROR Buffer mapped_framebuffer.cpp:230 Failed to mmap plane: Permission denied
2025-02-10 19:10:57.681   383-397   libcamera               and...amera.provider@2.5-service_64  E  ERROR Buffer mapped_framebuffer.cpp:230 Failed to mmap plane: Permission denied
2025-02-10 19:10:57.911   702-702   C3Dev-0-ReqQueu         cameraserver                         I  type=1400 audit(0.0:6526): avc:  denied  { open } for  path="/vendor/lib64/libgbm_mesa_wrapper.so" dev="mmcblk0p6" ino=730 scontext=u:r:cameraserver:s0 tcontext=u:object_r:vendor_file:s0 tclass=file permissive=1
2025-02-10 19:10:57.911   702-702   C3Dev-0-ReqQueu         cameraserver                         I  type=1400 audit(0.0:6527): avc:  denied  { getattr } for  path="/vendor/lib64/libgbm_mesa_wrapper.so" dev="mmcblk0p6" ino=730 scontext=u:r:cameraserver:s0 tcontext=u:object_r:vendor_file:s0 tclass=file permissive=1
2025-02-10 19:10:58.106   383-397   libcamera               and...amera.provider@2.5-service_64  E  ERROR Buffer mapped_framebuffer.cpp:230 Failed to mmap plane: Permission denied
2025-02-10 19:10:58.106   383-397   libcamera               and...amera.provider@2.5-service_64  E  ERROR V4L2 v4l2_videodevice.cpp:1960 /dev/video23[23:cap]: Failed to queue buffer 2: Invalid argument
2025-02-10 19:10:58.106   383-397   libcamera               and...amera.provider@2.5-service_64  E  ERROR RPISTREAM rpi_stream.cpp:276 Failed to queue buffer for ISP Output0
2025-02-10 19:10:58.106   383-397   libcamera               and...amera.provider@2.5-service_64  E  ERROR HAL camera_device.cpp:1142 '/base/axi/pcie@120000/rp1/i2c@88000/imx477@1a': Request 12970367432485081120 not successfully completed: 2
2025-02-10 19:10:59.180   383-397   libcamera               and...amera.provider@2.5-service_64  E  ERROR Buffer mapped_framebuffer.cpp:230 Failed to mmap plane: Permission denied
2025-02-10 19:10:59.283   383-397   libcamera               and...amera.provider@2.5-service_64  E  ERROR Buffer mapped_framebuffer.cpp:230 Failed to mmap plane: Permission denied
2025-02-10 19:10:59.584   383-397   libcamera               and...amera.provider@2.5-service_64  E  ERROR Buffer mapped_framebuffer.cpp:230 Failed to mmap plane: Permission denied
2025-02-10 19:10:59.884   383-397   libcamera               and...amera.provider@2.5-service_64  E  ERROR Buffer mapped_framebuffer.cpp:230 Failed to mmap plane: Permission denied
2025-02-10 19:11:00.181   383-397   libcamera               and...amera.provider@2.5-service_64  E  ERROR Buffer mapped_framebuffer.cpp:230 Failed to mmap plane: Permission denied
2025-02-10 19:11:00.480   383-397   libcamera               and...amera.provider@2.5-service_64  E  ERROR Buffer mapped_framebuffer.cpp:230 Failed to mmap plane: Permission denied
2025-02-10 19:11:00.559   816-2240  PointerChoreographer    system_server                        W  Invalid device id -1. Cannot set pointer icon.
2025-02-10 19:11:00.680   383-397   libcamera               and...amera.provider@2.5-service_64  E  ERROR Buffer mapped_framebuffer.cpp:230 Failed to mmap plane: Permission denied

KonstaT, 2025-02-27:
Yes, there is "various issues with CSI camera modules" as stated on my release pages. There's also some difference in CSI camera module functionality between Pi 4 and Pi 5 due to differences in hardware. Pull requests are welcome!

I only have the official v1 and v2 camera modules to test. I can't reproduce this on v2 module, full 8MP is used (https://github.com/raspberry-vanilla/android_external_libcamera/commit/0ef0b0ad1f2ea48a3cc4c9f405e0b22d8db6bba7).

Kernel and libcamera currently support for IMX219, IMX296, IMX477, IMX500, IMX708, and OV5647 camera sensors on my Android 15 releases (https://github.com/raspberry-vanilla/android_kernel_brcm_rpi/blob/android-15.0/arch/arm64/configs/android_rpi5_defconfig#L3848-L3925, https://github.com/raspberry-vanilla/android_device_brcm_rpi5/blob/android-15.0/device.mk#L109-L124 & https://github.com/raspberry-vanilla/android_device_brcm_rpi5/blob/android-15.0/camera/camera_hal.yaml). There's no support for Arducam camera sensors. I've just yesterday enabled the configurations for Arducam's IMX519 and OV64A40 but I haven't pushed the source yet.

danascape, 2025-03-02:
> Yes, there is "various issues with CSI camera modules" as stated on my release pages. There's also some difference in CSI camera module functionality between Pi 4 and Pi 5 due to differences in hardware. Pull requests are welcome!
> 
> I only have the official v1 and v2 camera modules to test. I can't reproduce the 1920x1080 capture on v2 module. Full 8MP is used (after [raspberry-vanilla/android_external_libcamera@0ef0b0a](https://github.com/raspberry-vanilla/android_external_libcamera/commit/0ef0b0ad1f2ea48a3cc4c9f405e0b22d8db6bba7)).
> 
> Kernel and libcamera currently support for IMX219, IMX296, IMX477, IMX500, IMX708, and OV5647 camera sensors on Android 15 (https://github.com/raspberry-vanilla/android_kernel_brcm_rpi/blob/android-15.0/arch/arm64/configs/android_rpi5_defconfig#L3848-L3925, https://github.com/raspberry-vanilla/android_device_brcm_rpi5/blob/android-15.0/device.mk#L109-L124 & https://github.com/raspberry-vanilla/android_device_brcm_rpi5/blob/android-15.0/camera/camera_hal.yaml). There's no support for Arducam camera sensors. I've just yesterday enabled the configurations for Arducam's IMX519 and OV64A40 but I haven't pushed the source yet.

Thank you for that, if possible can you provide me with some insights to help me with debugging the issues? I'd be happy to raise a PR if I am able to fix the issue

KonstaT, 2025-03-03:
> Thank you for that, if possible can you provide me with some insights to help me with debugging the issues?

You can change the libcamera log level here.
https://github.com/raspberry-vanilla/android_device_brcm_rpi5/blob/android-15.0/camera/android.hardware.camera.provider%402.5-service_64.rpi.rc#L9

Source code for the relevant parts.
libcamera: https://github.com/raspberry-vanilla/android_external_libcamera/tree/android-15.0
libpisp: https://github.com/raspberry-vanilla/android_external_libpisp/tree/android-15.0
minigbm: https://github.com/raspberry-vanilla/android_external_minigbm/tree/android-15.0

azlekov, 2025-04-11:
@danascape did you find any workaround or solution?

danascape, 2025-04-11:
> [@danascape](https://github.com/danascape) did you find any workaround or solution?

Nope nothing as of now, still working on it, We were able to point it to the driver, but still unsure with everything

azlekov, 2025-04-11:
I found this: https://github.com/raspberrypi/libcamera/issues/144 

From what I understand switching the format to NV21 should fix the issue, but I will test it in a while.

danascape, 2025-04-15:
> I found this: [raspberrypi/libcamera#144](https://github.com/raspberrypi/libcamera/issues/144)
> 
> From what I understand switching the format to NV21 should fix the issue, but I will test it in a while.

Do let us know if it works out, and feel free to make PR

azlekov, 2025-04-15:
We are trying to pivot out solution to other platform or OS as I do not have the knowledge to cherry-pick all changes and build the Android. Also the experiments are one year old, so I expect additional challenges on the way. But

> I found this: https://github.com/raspberrypi/libcamera/issues/144

Is a good starting point for someone who knows what and where to touch.

KonstaT, 2025-06-16:
`android-16.0` branch has an updated version based on libcamera v0.5.1. Switched from gralloc API to GraphicBufferAllocator as suggested (which doesn't actually fix anything besides move away from using a deprecated API).

There's some obvious issue configuring the camera on rpi IPAs. I've omitted one previous HACK (https://github.com/raspberry-vanilla/android_external_libcamera/commit/0ef0b0ad1f2ea48a3cc4c9f405e0b22d8db6bba7) as it causes capture to fail on Pi 5. With camera module v2 it now captures 1437x1080 image vs. the full resolution. At least it captures something and the behaviour is consistent between Pi 4 and Pi 5.

Current status using the AOSP camera application:
- Streaming (camera preview) works but can freeze after some time
- Capturing photo works on low resolution
- Capturing video doesn't work

There are two major things in TODO:
- Move away from the meson build system HACK and adapt soong blueprint makefiles (`android-16.0` already shows the first signs of deprecating the legacy Android.mk build system)
- Adapt AIDL camera HAL (`android.hardware.camera.provider@2.5::ICameraProvider/legacy/0` has been deprecated in recent Android versions, work on BayLibre TI libcamera fork can be used as a reference https://gitlab.baylibre.com/baylibre/ti/android/aosp/external/libcamera)

In any case, the above two require significant development effort and will introduce some maintenance issues moving forward. Other option is to drop CSI camera module support from Raspberry Vanilla project altogether if no one steps up to fix the actual issues with libcamera on AOSP (Pi IPA/pipeline specifically).

azlekov, 2025-06-27:
Is there a build which I can try? What is the current status quo with the cameras overall?

KonstaT, 2025-06-28:
> Is there a build which I can try? What is the current status quo with the cameras overall?

There's a status report just few posts up that matches my latest AOSP16 releases. For any further updates you would need to compile yourself for now.

Changing the build system or plumbing on the camera stack doesn't make any change that you should notice as en end user (actually, it would only be a positive sign if you don't notice any difference). libcamera implementation itself is still the same and still has the same issues.