raspberry-vanilla/android_local_manifest #188: Enabling a2dp sink

RobertK0, 2025-03-01:
Moving the discussion here as per emails.

Not exactly sure which of these are necessary for which features, but at least for enabling a2dp sink, what should be changed is: 
[bluetooth package config.xml, a2dp_sink_automatically_request_audio_focus](https://github.com/raspberry-vanilla/android_packages_modules_Bluetooth/blob/5e28636043934afbc06d137867c4796c17ac7149/android/app/res/values/config.xml#L111C74-L112C1), set value to true (I reckon this should probably not affect anything outside of A2dpSinkService)

And the rest are build.props which can be changed at any point by user, and can be found [here](https://android.googlesource.com/platform/packages/services/Car/+/refs/heads/main/car_product/properties/bluetooth.prop), and they are: 

bluetooth.device.class_of_device=38,4,32 #this specifically defines device as an audio receiver, in this case, a car multimedia set

\#The Bluetooth profiles that cars expect to have enabled. 
bluetooth.profile.a2dp.sink.enabled=true
bluetooth.profile.avrcp.controller.enabled=true
bluetooth.profile.a2dp.source.enabled=true
bluetooth.profile.avrcp.target.enabled=true
bluetooth.profile.gatt.enabled=true
bluetooth.profile.hfp.hf.enabled=true
bluetooth.profile.map.client.enabled=true
bluetooth.profile.pan.nap.enabled=true
bluetooth.profile.pan.panu.enabled=true
bluetooth.profile.pbap.client.enabled=true
bluetooth.gatt.check_encrypted_link.enabled=false

\# Enable art cover for a2dp source
persist.bluetooth.avrcpversion=avrcp16
persist.bluetooth.avrcpcontrolversion=avrcp16

(Don't know if both sink and source can be enabled, so I tested it out without, and it worked)

closed 2025-03-01 by KonstaT

KonstaT, 2025-03-01:
Looked into this earlier today and looks like it doesn't need to be discussed further. ;)

RobertK0, 2025-03-01:
Nice. 
Does this mean that this will also be included in your next aosp/lineage build, as I assume they are based on raspberry-vanilla?

KonstaT, 2025-03-01:
Yes.

RobertK0, 2025-03-01:
Gotcha, any ETA on that by any chance?

KonstaT, 2025-03-01:
You can compile this any time you like. :)

There's no ETAs for any releases (nor should you ask for one). QPR2 is next month(?) and I don't know what issues it might bring.

RobertK0, 2025-03-01:
Oh I got it compiled and running for a couple of days now, though I prefer the additional features of your builds, hence the question.
  
Anyway, no rush in that case, fingers crossed everything goes smooth!