raspberry-vanilla/android_local_manifest #189: Building libcamera

a2tq, 2025-03-17:
I'm trying to compile Android for Pi 5 (`aosp_rpi5_car-bp1a-userdebug` and `aosp_rpi5-bp1a-userdebug` targets) but I have a problem with compilation libcamera (exactly `libcamera-base.so`). I get the following error:
```
ld.lld: error: /usr/lib/libatomic.so is incompatible with aarch64linux
```
It seems that meson took the library from the host system instead of... Well, precisely, I don't know from which module this library should come. I don't have such library in the workspace.
The command (after building with `make -k`):
```bash
find -name "libatomic*.so*"`
```
returns:
```
./android/prebuilts/gcc/linux-x86/host/x86_64-linux-glibc2.17-4.8/x86_64-linux/lib64/libatomic.so.1
./android/prebuilts/gcc/linux-x86/host/x86_64-linux-glibc2.17-4.8/x86_64-linux/lib64/libatomic.so
./android/prebuilts/gcc/linux-x86/host/x86_64-linux-glibc2.17-4.8/x86_64-linux/lib64/libatomic.so.1.0.0
./android/prebuilts/gcc/linux-x86/host/x86_64-linux-glibc2.17-4.8/x86_64-linux/lib32/libatomic.so.1
./android/prebuilts/gcc/linux-x86/host/x86_64-linux-glibc2.17-4.8/x86_64-linux/lib32/libatomic.so
./android/prebuilts/gcc/linux-x86/host/x86_64-linux-glibc2.17-4.8/x86_64-linux/lib32/libatomic.so.1.0.0
./android/out/soong/.intermediates/external/rust/android-crates-io/crates/atomic/libatomic_rust/android_arm64_armv8-a_cortex-a76_dylib/libatomic_rust.dylib.so
./android/out/soong/.intermediates/external/rust/android-crates-io/crates/atomic/libatomic_rust/android_arm64_armv8-a_cortex-a76_dylib/libatomic_rust.dylib.so.d
./android/out/soong/.intermediates/external/rust/android-crates-io/crates/atomic/libatomic_rust/android_arm64_armv8-a_cortex-a76_dylib/unstripped/libatomic_rust.dylib.so
./android/out/soong/.intermediates/external/rust/android-crates-io/crates/atomic/libatomic_rust/android_arm64_armv8-a_cortex-a76_dylib/unstripped/libatomic_rust.dylib.so.d
./android/out/soong/.intermediates/external/rust/android-crates-io/crates/atomic/libatomic_rust/android_arm64_armv8-a_cortex-a76_dylib/unstripped/libatomic_rust.dylib.so.d.raw
./android/out/target/product/rpi5/obj/DYLIB_LIBRARIES/libatomic_rust_intermediates/libatomic_rust.dylib.so
./android/out/target/product/rpi5/obj/PACKAGING/elf_symbol_mapping_intermediates/system/lib64/libatomic_rust.dylib.so.textproto
./android/out/target/product/rpi5/system/lib64/libatomic_rust.dylib.so
./android/out/target/product/rpi5/symbols/system/lib64/libatomic_rust.dylib.so
```
These from `prebuilts` are in host system architecture.
What can I check more? From which module should `libatomic` come from?

KonstaT, 2025-03-18:
What Linux distro are you using to build? Have you established the build environment following the instructions?

Meson build system that is used for building e.g. libcamera is actually a hack and not how libraries should be built on Android. Generally AOSP tree provides the necessary build tools but with the meson build system, the tools you have installed on your host machine also make a difference.

> From which module should `libatomic` come from?

Atomic functions should be part of the compiler AFAICT.
https://android.googlesource.com/platform/build/soong/+/c2018e2ed4ef5c8d9eeb23d563fbe5791f9992d5%5E%21/#F0

Likely it shouldn't link a library at all and most definitely not one from the host system. So, something strange with your build environment.

a2tq, 2025-03-18:
I'm on Arch Linux. I believe I have installed all necessary tools.
Can you provide me the log from your build of libcamera? I would compare it to the [mine](https://github.com/user-attachments/files/19325804/verbose.log).

KonstaT, 2025-03-18:
> I'm on Arch Linux.

Ubuntu 22.04 is the recommended build environment.

Mine has
```
ld.lld: error: unable to find library -latomic
...
Library atomic found: NO
```
vs. in yours
```
Library atomic found: YES
```

IDK, why. Maybe check `pkg-config --list-all` if there's anything atomic related (nothing on mine) and figure out what provides it.

a2tq, 2025-03-25:
I see nothing related to libatomic in `pkg-config --list-all`.
Perhaps `aosp_cross` files are wrongly generated. Here is mine from 
[out/target/product/rpi5/obj/MESON_LIBCAMERA_GEN/aosp_cross](https://github.com/user-attachments/files/19454407/aosp_cross.txt).

Temporary I bypassed the problem by setting:
```
libatomic = dependency('', required : false)
```
in `external/libcamera/src/libcamera/meson.build`.

KonstaT, 2025-04-10:
Closing as there hasn't been any update in weeks. Ubuntu 22.04 LTS is the recommended build environment and a workaround for Arch Linux was discovered.

closed 2025-04-10 by KonstaT