raspberry-vanilla/android_local_manifest #190: Waveshare 10.1" 1280x800 display locked to 30hz

ftab, 2025-03-19:
mSupportedModesByDisplay: 0 -> [{id=1, width=1280, height=800, fps=60.02573, vsync=60.02573, synthetic=false, alternativeRefreshRates=[], supportedHdrTypes=[]}]

Yet...

renderFrameRate 30.012865

[WaveshareDisplay_Android_dumpsys_display.txt](https://github.com/user-attachments/files/19320984/WaveshareDisplay_Android_dumpsys_display.txt)

[logcat.txt](https://github.com/user-attachments/files/19346926/logcat.txt)

Tried with Raspberry Vanilla (Android Automotive) and your AOSP image, same deal

Hooked up to raspberry pi5 dsi1 (also tried dsi0). The driver is set up for 2 lanes according to https://github.com/raspberrypi/linux/blob/0553897d77e849a86e836ddf1e0c0dbbd8e64f83/drivers/gpu/drm/panel/panel-waveshare-dsi.c#L164

An HDMI display successfully ran at 60fps when I tried that

Any ideas how I can get it to run at 60fps on the Waveshare display too?

Waveshare support suggested it might be software settings (SurfaceFlinger) but I'm having a hard time finding any that pertain to render framerate.

KonstaT, 2025-03-19:
You could try setting min_refresh_rate. Someone reported this fixed similar issue on Waveshare 4" DSI display.

In your case e.g.
```
adb root
adb shell
settings put system min_refresh_rate 60.02
```

ftab, 2025-03-19:
```
$ adb remount
Device must be bootloader unlocked
```

But the first one seems to have done the trick, thanks!

```
renderFrameRate 60.02573
```

Now I'm off to figure out how to bake that setting into the image.

closed 2025-03-19 by ftab

KonstaT, 2025-03-19:
> Device must be bootloader unlocked

https://github.com/raspberry-vanilla/android_local_manifest/issues/47#issuecomment-1829888921

or
```
adb root
adb shell
mount -o rw,remount /vendor
```

On Pi 5 DSI-0, DSI-1, DPI, and VEC all have independent cards. Actually this might be related to that as the `logcat` was missing the resolutions hwcomposer should get from the display it reports as supported.

Try adding `vendor.hwc.drm.device=/dev/dri/card2` to /vendor/build.prop (i.e. you can also add it here https://github.com/raspberry-vanilla/android_device_brcm_rpi5/blob/android-15.0/vendor.prop and rebuild).

KonstaT, 2025-03-26:
Does setting `vendor.hwc.drm.device=/dev/dri/card2` resolve the issue or do you still need to `settings put system min_refresh_rate ...`?

ftab, 2025-03-26:
Looks like I still need that setting