raspberry-vanilla/android_local_manifest #191: Build environment question.

ama-784, 2025-03-21:
> ama@AMA:~/ras-vanil$ make bootimage systemimage vendorimage -j1
> 17:16:41 Build sandboxing disabled due to nsjail error.
> 17:16:42 ************************************************************
> 17:16:42 You are building on a machine with 15.5GB of RAM
> 17:16:42 
> 17:16:42 The minimum required amount of free memory is around 16GB,
> 17:16:42 and even with that, some configurations may not work.
> 17:16:42 
> 17:16:42 If you run into segfaults or other errors, try reducing your
> 17:16:42 -j value.
> 17:16:42 ************************************************************
> ============================================
> PLATFORM_VERSION_CODENAME=REL
> PLATFORM_VERSION=15
> TARGET_PRODUCT=aosp_rpi5
> TARGET_BUILD_VARIANT=userdebug
> TARGET_ARCH=arm64
> TARGET_ARCH_VARIANT=armv8-a
> TARGET_CPU_VARIANT=cortex-a76
> TARGET_2ND_ARCH=arm
> TARGET_2ND_ARCH_VARIANT=armv7-a-neon
> TARGET_2ND_CPU_VARIANT=generic
> HOST_OS=linux
> HOST_OS_EXTRA=Linux-6.11.0-19-generic-x86_64-Ubuntu-24.04.2-LTS
> HOST_CROSS_OS=windows
> BUILD_ID=BP1A.250305.019
> OUT_DIR=out
> ============================================
> [100% 2/2] analyzing Android.bp files and generating ninja file at out/soong/bui
> FAILED: out/soong/build.aosp_rpi5.ninja
> cd "$(dirname "out/host/linux-x86/bin/soong_build")" && BUILDER="$PWD/$(basename
>  "out/host/linux-x86/bin/soong_build")" && cd / && env -i  "$BUILDER"     --top 
> "$TOP"     --soong_out "out/soong"     --out "out"     --soong_variables out/soo
> ng/soong.aosp_rpi5.variables -o out/soong/build.aosp_rpi5.ninja -l out/.module_p
> aths/Android.bp.list --available_env out/soong/soong.environment.available --use
> d_env out/soong/soong.environment.used.aosp_rpi5.build Android.bp
> Warning: Module 'androidx.wear_wear' depends on non-existing optional_uses_libs 
> 'wear-sdk'
> Warning: Module 'androidx.wear.compose_compose-foundation' depends on non-existi
> ng optional_uses_libs 'wear-sdk'
> Killed
> 17:18:26 soong bootstrap failed with: exit status 1
> ninja: build stopped: subcommand failed.
> 
> #### failed to build some targets (01:47 (mm:ss)) ####

Hi @KonstaT I cannot build android because of above issue.
#128 #169 was that problem but it does not have solution.

As a development computer, I am using 
i7-7700
16GB RAM
no graphic card
and using ZRAM

Is there any way to get build logs? Please provide for me if needed.
I do not want to change my dev computer device setups.

KonstaT, 2025-03-22:
As mentioned in the hardware requirements, you'll need "minimum of 64 GB of RAM".
https://source.android.com/docs/setup/start/requirements#hardware-requirements

You can get away with 32GB and a large swap file but 16GB is nowhere near enough. zRAM is compressed memory vs. swap file.

`Killed` doesn't really provide any useful information as there's no error message in the output.

ama-784, 2025-03-23:
Ok thank you for the response and editing the original issue.
I will update this issue after my 32gB(16*2) RAM is delivered for my workstation.

I upload the log files you mentioned, but as my android dev is the first time I am trying so could you tell me which one I have to look?

[BuildErros.tar.gz](https://github.com/user-attachments/files/19407283/BuildErros.tar.gz)

KonstaT, 2025-03-23:
I guess all of them depending on what/how fails. Ones that have a number are from previous builds.

error.log
```
FAILED: analyzing Android.bp files and generating ninja file at out/soong/build.aosp_rpi5.ninja
...
Killed
```

soong.log
```
2025/03/21 17:16:43.607840 build/soong/ui/logger/logger.go:290: "soong bootstrap" executing "prebuilts/build-tools/linux-x86/bin/ninja" [prebuilts/build-tools/linux-x86/bin/ninja -d keepdepfile -d stats -o usesphonyoutputs=yes -o preremoveoutputs=yes -w dupbuild=err -w outputdir=err -w missingoutfile=err -j 1 --frontend_file out/.ninja_fifo -f out/soong/bootstrap.ninja out/soong/build.aosp_rpi5.ninja]
2025/03/21 17:18:26.944110 build/soong/ui/logger/logger.go:290: "soong bootstrap" finished with exit code 1 (1m43.336s real, 8m8.259s user, 2m2.449s system, 14734MB maxrss)
2025/03/21 17:18:26.945198 build/soong/ui/build/exec.go:127: soong bootstrap failed with: exit status 1
```

Which still isn't very useful. So I'd debug the
```
Build sandboxing disabled due to nsjail error.
```

What Linux distro are you using? Make sure you've followed the instructions on how to setup the build environment. Ubuntu 22.04 is recommended.

ama-784, 2025-03-23:
@KonstaT 
Linux 22.04 of course, I just started to use my dev PC.
[image]

BTW,
How can I debug 'Build sandboxing disabled due to nsjail error'? 
I have no idea for doing that one.

KonstaT, 2025-03-23:
> How can I debug 'Build sandboxing disabled due to nsjail error'?

Trade secrets but generally people copy&paste the error message to Google search if it's isn't obvious. :P

Maybe worth trying this time as well. You might even find it on AOSP and Ubuntu bug trackers.

ama-784, 2025-03-26:
Hi, there's an update.
I bought 16GB RAM x2 and working on.

And apologize for above logs, I was overlooking the error when I started lunch.
When I started 
> lunch aosp_rpi5-bp1a-userdebug
>

after make clean,
I see below message.
> Build sandboxing disabled due to nsjail error
>

I guess build environment setting is not well completed.
I see there are multiple solutions for multiple causes when I google it, and I am trying to modify the soong script.
I will figure it out and update.

KonstaT, 2025-03-26:
Like mentioned, Ubuntu 22.04 LTS is the recommended build environment. There's also reports with issues installing the required python packages on Ubuntu 24.04 LTS.

abhisarma7, 2025-03-26:
Hey, Just for info, My 2013 mac pro runs ubuntu 24.04 with 32 gb ram and the build works without issues and I've built the image multiple times without errors. What I've observed is that the build gets killed due to swap mem reaching its max. There are two options to resolve this, kill the cpp-tools process after the build gets failed and you would see that the swap mem and ram mem goes to the min or else just restart the system and the build gets successful. 

Hope this helps !

abhisarma7, 2025-03-26:
Build sandboxing disabled due to nsjail error

This is common and I've seen it multiple times but finally, the build gets successfully.

The process gets killed if the swap or ram mem is full , this only happens during the step analysing the android.bp files, not for the .mk files. So if any changes made to any of the android.bp files, close all apps except the terminal and after that step you can see that ram and swap mem util goes down so much during the step where it analyses android.mk files.

ama-784, 2025-03-26:
Hi @abhisarma7 Thanks for info. 
It’s a little bit weird because I only tried to setup my pc for this aosp build, and problem occurs just when I boot up and try build. So memory corruption or loading error might not be the major problem of this issue.(Or maybe I should study about ram swaps)

Firstly, I will try to install Ubuntu 22.04 and try build AOSP on my dev PC.
and secondly, I will investigate soong build configures;
I will figure out; Will the settings regarding aarch64/ARM64 cause nsjail build issue? 

One might be possible reason that I fail, I used xrdp to get into my dev PC, I will try ssh after installing Ubuntu 22.04

abhisarma7, 2025-03-27:
I dont think you need to install ubuntu 24.04, as you can see in my screenshot - 

[image]

In the screenshot it says Build sand-boxing disabled due to nsjail error, but the build is successful. from what I've remember, in latest Ubuntu there is a concept where the executables(i.e in this instance build executable) are run with root privileges and another version of the same executable would run in a sandboxed version with user privileges (this is for some security reasons and IDK why), and the one which is sand boxed would always fail if the executable would try to access resources which it doesnt have the previliages when ran with user access. I think it might be related to that but I cant say for sure its because of that.

Overall, That is not the reason for it getting killed. Do one thing, run your build but on the side open system monitor, observe in the graph whether the build is getting killed when the ram and swap memory is full( it would shown in the graph). If it is, see if cpp-tools service is using max ram or max disk util, if so kill it and you would see the utilization going down and try to rebuild the image.

ama-784, 2025-03-27:
I will try that after a few hours when I leave from work :) Thanks for information, and will leave note with my status tagging you.

ama-784, 2025-03-27:
On my dev PC, I installed Ubuntu 22.04 as @KonstaT mentioned. As this repository was made and configured with 22.04, I just accepted it.
Rather than using xrdp for remote connection, I installed openssh so that I can connect to it remotely.(Maybe it will require less resource)

As @abhisarma7 told me, I will keep memory clean after doing some stuff. Let's see if these two works for me.

* If it fails again, I will try 1. swapoff original swap memory(if exist) and try, if 1 fails, I will try 2: install ZRAM and try some.
I will update as my situation goes after repo sync. (If I am not drunk a lot.. 🥇  )

ama-784, 2025-03-27:
[image]
It seems NSJAIL error does not come again.

[image]
Build seems to be OK, too.

Thanks everyone, I will update when image build is finished, and if there isn't any error, I will close the issue.

ama-784, 2025-03-28:
Status update :
on lunch aosp_rpi5-bp1a-userdebug :

1.
[image]
I don't know why "Wondering whether to use user, userdebug, or eng?" appears even though I put userdebug above as a guide to make an image. Target CPU is well set to cortex a76 and other options are okay, but why it appears? Anyway when I proceed to next,

2.

[image]
This problem still appears.

Need some help here @KonstaT  @abhisarma7 

Ubuntu 22.04 is well installed,
[image] 

From @abhisarma7 's comment I swapped off all, but still happens

[image]

Memory info : 

[image]

ama-784, 2025-03-28:
Status update :

[image]
I tried to build an image while monitoring the RAM state.
From total 31GB, build process remained only 180MB~190 free RAM space for 2 mins, and the build process suddenly dies with "Killed"

I have remaining 8+8 GB RAM, I will put it into my dev PC and try again.(Hope nothing goes wrong..)

ama-784, 2025-03-28:
Build is going on after I set up 16 + 8 + 16 + 8 GB (Total 48GB) Ram.

[image]

Now I know that this issue was triggered by small RAM size.

I will close the issue, thanks for helping me @KonstaT , @abhisarma7,
hope to have a chance to help you guys :)

closed 2025-03-28 by ama-784