raspberry-vanilla/android_local_manifest #194: [Rpi5] please allow under-clocking

Technologov, 2025-04-06:
Default clock of 2400 MHz on Rpi5. (Rasp Pi 5 settings)

The problem is:
If I am using some 3rd part 3 Amper charging, it works unstable.

Instead of allowing only over-clocking up to 2.9 GHz, please allow under-clocking down to 1600 MHz. This would make Rpi5 much more stable with non-official chargers.

closed 2025-04-07 by KonstaT

Technologov, 2025-04-07:
> Closing as Raspberry Vanilla project doesn't provide any overclocking
> options.
>
> You can refer to the official Raspberry Pi documentation and
> over/underclock your device as you please.
>
> https://www.raspberrypi.com/documentation/computers/config_txt.html#overclocking-options
>
> Appropriate solution is to get proper power supply for your device of
> course.

Your own build provides overclocking to 2.9 GHz.

-Alexey Eromenko "Technologov"

KonstaT, 2025-04-07:
And no one is forcing to use that either.

It's nothing but a more user friendly way to do exactly what is described in the documentation I linked above.

Technologov, 2025-04-07:
> And no one is forcing you to use that either.

Since over-clocking is actually supported, it would be nice to support
under-clocking also.
It would make Raspberry Pi a lot more stable.

-Alexey Eromenko "Technologov"

KonstaT, 2025-04-08:
Again, Raspberry Vanilla project doesn't provide any overclocking options. It requires no support whatsoever from me for you to be able to over/underclock your device as you please. I can absolutely guarantee there will be no underclocking options in my future releases.

Using appropriate power supply would make your Raspberry Pi a lot more stable.