raspberry-vanilla/android_local_manifest #195: Antutu 3D benchmark crash

abhisarma7, 2025-04-09:
Hey @KonstaT,

It might sound silly, but for benchmarking purposes, I ran Antutu Benchmark 3D test, and for the gpu test, It was crashing. It says that it found the v3d 7.1.7.0 driver( Vulkan driver) but for tests such as Seasons-Vulkan and Coastline-Vulkan, it says that its not supported. 

It it because of the vulkan driver in the ROM or just the app is glitching??

abhisarma7, 2025-04-10:
Just checked with geekbench 6, the same thing, GPU tests arent working but for some reason geekbench 5 worked. 
Since, both antutu and geekbench are crashing, is it a vulkan service issue?

KonstaT, 2025-04-10:
> It it because of the vulkan driver in the ROM or just the app is glitching??

> Since, both antutu and geekbench are crashing, is it a vulkan service issue?

That's what the logs are for so there's no need to guess. Please attach the appropriate logs (`logcat` in this case).

abhisarma7, 2025-04-10:
[geekbench_crash_log.txt](https://github.com/user-attachments/files/19690377/geekbench_crash_log.txt)

I've uploaded the entire logcat. I ran the app at 18:31(for time reference), 

I've noticed this error - 
`04-10 18:31:20.586  4401  4401 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
04-10 18:31:20.586  4401  4401 F DEBUG   : Build fingerprint: 'Aegis/aosp_rpi5/rpi5:15/AP4A.250205.002/eng.androi:userdebug/test-keys'
04-10 18:31:20.586  4401  4401 F DEBUG   : Revision: '0'
04-10 18:31:20.586  4401  4401 F DEBUG   : ABI: 'arm64'
04-10 18:31:20.586  4401  4401 F DEBUG   : Timestamp: 2025-04-10 18:31:20.255263542+0000
04-10 18:31:20.586  4401  4401 F DEBUG   : Process uptime: 88s
04-10 18:31:20.586  4401  4401 F DEBUG   : Cmdline: com.primatelabs.geekbench6
04-10 18:31:20.586  4401  4401 F DEBUG   : pid: 4237, tid: 4387, name: AsyncTask #1  >>> com.primatelabs.geekbench6 <<<
04-10 18:31:20.586  4401  4401 F DEBUG   : uid: 10101
04-10 18:31:20.586  4401  4401 F DEBUG   : tagged_addr_ctrl: 0000000000000001 (PR_TAGGED_ADDR_ENABLE)
04-10 18:31:20.586  4401  4401 F DEBUG   : signal 6 (SIGABRT), code -1 (SI_QUEUE), fault addr --------
04-10 18:31:20.586  4401  4401 F DEBUG   : Abort message: 'terminating with uncaught exception of type std::runtime_error: failed to find suitable memory type'
04-10 18:31:20.586  4401  4401 F DEBUG   :     x0  0000000000000000  x1  0000000000001123  x2  0000000000000006  x3  0000007b08f53530
04-10 18:31:20.586  4401  4401 F DEBUG   :     x4  736f646277641f73  x5  736f646277641f73  x6  736f646277641f73  x7  7f7f7f7f7f7f7f7f
04-10 18:31:20.586  4401  4401 F DEBUG   :     x8  00000000000000f0  x9  7fad1f4f1d6912aa  x10 0000000000000001  x11 0000007e57ef7c30
04-10 18:31:20.586  4401  4401 F DEBUG   :     x12 000000007fffffff  x13 0000000000059caa  x14 000000157b3f2bf0  x15 000000001284bda1
04-10 18:31:20.586  4401  4401 F DEBUG   :     x16 0000007e57f62038  x17 0000007e57f4b650  x18 0000007a6c568000  x19 000000000000108d
04-10 18:31:20.586  4401  4401 F DEBUG   :     x20 0000000000001123  x21 00000000ffffffff  x22 0000007b08f53660  x23 0000007b08f536a0
04-10 18:31:20.586  4401  4401 F DEBUG   :     x24 0000007b08f53780  x25 0000007b08f536c8  x26 0000000000000000  x27 0000007e51174b58
04-10 18:31:20.586  4401  4401 F DEBUG   :     x28 0000000000000009  x29 0000007b08f535b0
04-10 18:31:20.586  4401  4401 F DEBUG   :     lr  0000007e57ee0d58  sp  0000007b08f53530  pc  0000007e57ee0d7c  pst 0000000000001000
04-10 18:31:20.586  4401  4401 F DEBUG   : 37 total frames
04-10 18:31:20.586  4401  4401 F DEBUG   : backtrace:
04-10 18:31:20.586  4401  4401 F DEBUG   :       #00 pc 000000000005dd7c  /apex/com.android.runtime/lib64/bionic/libc.so (abort+156) (BuildId: b7e90ed7765733750bff48980ee1ba2d)
04-10 18:31:20.586  4401  4401 F DEBUG   :       #01 pc 00000000000d1fcc  /data/app/~~08CeBEy3l24CxVCVTFtiog==/com.primatelabs.geekbench6-GrS6Te0eWZ0_uNC6hbBxqA==/split_config.arm64_v8a.apk!libc++_shared.so (offset 0x1000) (BuildId: 7dc5f79124fbc1a486506d25783650b07e98a111)
04-10 18:31:20.586  4401  4401 F DEBUG   :       #02 pc 00000000000d2180  /data/app/~~08CeBEy3l24CxVCVTFtiog==/com.primatelabs.geekbench6-GrS6Te0eWZ0_uNC6hbBxqA==/split_config.arm64_v8a.apk!libc++_shared.so (offset 0x1000) (BuildId: 7dc5f79124fbc1a486506d25783650b07e98a111)
04-10 18:31:20.586  4401  4401 F DEBUG   :       #03 pc 00000000000e6e80  /data/app/~~08CeBEy3l24CxVCVTFtiog==/com.primatelabs.geekbench6-GrS6Te0eWZ0_uNC6hbBxqA==/split_config.arm64_v8a.apk!libc++_shared.so (offset 0x1000) (BuildId: 7dc5f79124fbc1a486506d25783650b07e98a111)
04-10 18:31:20.586  4401  4401 F DEBUG   :       #04 pc 00000000000e6514  /data/app/~~08CeBEy3l24CxVCVTFtiog==/com.primatelabs.geekbench6-GrS6Te0eWZ0_uNC6hbBxqA==/split_config.arm64_v8a.apk!libc++_shared.so (offset 0x1000) (BuildId: 7dc5f79124fbc1a486506d25783650b07e98a111)
04-10 18:31:20.586  4401  4401 F DEBUG   :       #05 pc 00000000000e646c  /data/app/~~08CeBEy3l24CxVCVTFtiog==/com.primatelabs.geekbench6-GrS6Te0eWZ0_uNC6hbBxqA==/split_config.arm64_v8a.apk!libc++_shared.so (offset 0x1000) (__cxa_throw+124) (BuildId: 7dc5f79124fbc1a486506d25783650b07e98a111)
04-10 18:31:20.586  4401  4401 F DEBUG   :       #06 pc 0000000001339084  /data/app/~~08CeBEy3l24CxVCVTFtiog==/com.primatelabs.geekbench6-GrS6Te0eWZ0_uNC6hbBxqA==/split_config.arm64_v8a.apk!libgeekbench.so (offset 0xfc000) (BuildId: 807066d686bb267d940326e2a750488fbca9df05)
04-10 18:31:20.586  4401  4401 F DEBUG   :       #07 pc 00000000013394c0  /data/app/~~08CeBEy3l24CxVCVTFtiog==/com.primatelabs.geekbench6-GrS6Te0eWZ0_uNC6hbBxqA==/split_config.arm64_v8a.apk!libgeekbench.so (offset 0xfc000) (BuildId: 807066d686bb267d940326e2a750488fbca9df05)
04-10 18:31:20.586  4401  4401 F DEBUG   :       #08 pc 0000000001331418  /data/app/~~08CeBEy3l24CxVCVTFtiog==/com.primatelabs.geekbench6-GrS6Te0eWZ0_uNC6hbBxqA==/split_config.arm64_v8a.apk!libgeekbench.so (offset 0xfc000) (BuildId: 807066d686bb267d940326e2a750488fbca9df05)
04-10 18:31:20.586  4401  4401 F DEBUG   :       #09 pc 0000000001330dc4  /data/app/~~08CeBEy3l24CxVCVTFtiog==/com.primatelabs.geekbench6-GrS6Te0eWZ0_uNC6hbBxqA==/split_config.arm64_v8a.apk!libgeekbench.so (offset 0xfc000) (BuildId: 807066d686bb267d940326e2a750488fbca9df05)
04-10 18:31:20.587  4401  4401 F DEBUG   :       #10 pc 00000000012eb6fc  /data/app/~~08CeBEy3l24CxVCVTFtiog==/com.primatelabs.geekbench6-GrS6Te0eWZ0_uNC6hbBxqA==/split_config.arm64_v8a.apk!libgeekbench.so (offset 0xfc00 [... cut ...]
04-10 18:31:20.724   759  1526 I WindowManager: WIN DEATH: Window{dd88114 u0 Geekbench 6 EXITING}
04-10 18:31:20.724   759  1526 I WindowManager: WIN DEATH: Window{23a099d u0 com.primatelabs.geekbench6/com.primatelabs.geekbench.HomeActivity}
04-10 18:31:20.725   374   374 I Zygote  : Process 4237 exited due to signal 6 (Aborted)`

Mainly this line - 
`04-10 18:31:20.586  4401  4401 F DEBUG   : Abort message: 'terminating with uncaught exception of type std::runtime_error: failed to find suitable memory type'`

abhisarma7, 2025-04-10:
[geekbench_crash_log.txt](https://github.com/user-attachments/files/19690422/geekbench_crash_log.txt)

I've uploaded the entire log, and I ran the app at 18:31 (for time reference).

I've noticed this  line - 

> 04-10 18:31:20.586  4401  4401 F DEBUG   : Abort message: 'terminating with uncaught exception of type std::runtime_error: failed to find suitable memory type'

any clue on whats causing this??

KonstaT, 2025-04-11:
No idea TBH. You would probably need to ask Mesa/Vulkan developers. Some Vulkan features Android expects are not implemented on V3DV (https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/32319) but I doubt it's related to this. Attached full tombstone as I can reproduce this on Geekbench GPU test. Feel free to investigate.

[tombstone_00.txt](https://github.com/user-attachments/files/19703890/tombstone_00.txt)

abhisarma7, 2025-04-12:
The geekbench 5 works fine but not 6. In the gitlab repo you've shared, I saw somewhere that they need to update the current vulkan api impl methods to the latest api impl methods ( which geekbench 6 might be using), Thus might be the reason geekbench 5 works but not 6.

Anyways, I'll debug more and figure out the exact issue.