raspberry-vanilla/android_local_manifest #196: Fails to finish booting (stuck on boot animation)

ftab, 2025-04-10:
[logcat (1).log](https://github.com/user-attachments/files/19696518/logcat.1.log)
[logcat.log](https://github.com/user-attachments/files/19696517/logcat.log)

These logcats were collected from our custom build that is a lightly modified Raspberry Vanilla, there are probably some irrelevant crashes there (e.g. GNSS service I still haven't figured out).

I looked over them but don't know enough to really know what to look for vs. a working build (such as [logcat-dennis-2.log](https://github.com/user-attachments/files/19696664/logcat-dennis-2.log) ).

I built regular Raspberry Vanilla aosp_rpi5_car-bp1a-userdebug from a fresh sync as of today and it does the same thing--stuck at the boot animation. I am currently unable to get my Pi to give me adb over the USB-C cable so I can't get a fresh logcat from your original source. I will keep trying.

I suspect something that's changed within the last week or two has caused the failure, because I had builds that were working recently until today when I blew away the out/ directory, synced, and built from scratch.

Unless you already know what happened, tomorrow I will start reverting things from the last week or two until it starts working again. I will see if I can narrow it down.

Besides that, any ideas on what else I can try to get more diagnostics or figure this out?

KonstaT, 2025-04-11:
> I built regular Raspberry Vanilla aosp_rpi5_car-bp1a-userdebug from a fresh sync as of today...

That's the only thing I'm interested in scope of this project.

I can only see GNSS errors in your logs and I can't say it's irrelevant. FYI AOSP 15 build system has a bug that your changes to 'manifest.xml' don't get replicated to `/vendor/etc/vintf/manifest.xml`unless you `make clean` or do tricks with DEVICE_MANIFEST_FILE in BoardConfig.mk.

I've built `aosp_rpi5_car` on Android 15 QPR2 and I doubt any recent changes have broken it. I can check later when I get around building it.

KonstaT, 2025-04-11:
Closing as there's no issue booting `aosp_rpi5_car` on `android-15.0.0_r26` (or `android-15.0.0_r20` I had previously built).

Rather look into reverting your own changes and/or configuring the GNSS service properly (i.e. check your `/vendor/etc/vintf/manifest.xml`).

closed 2025-04-11 by KonstaT