raspberry-vanilla/android_local_manifest #197: AOSP 15 Build Fails on Ubuntu 18: 'meson' is not allowed to be used

bhaskarreddyv123, 2025-04-11:
Hii,

I am facing problem that described below
 
Environment:
- AOSP version: Android 15 (android-15.0.0_r26)
- Host OS: Ubuntu 18.04.6 LTS
- Target device: Raspberry Pi 4 (`aosp_rpi4_car`)

While building the AOSP 15 what i given above it throws an error

###Steps to Reproduce:
1. repo init -u https://android.googlesource.com/platform/manifest -b android-15.0.0_r26 --depth=1
2. curl -o .repo/local_manifests/manifest_brcm_rpi.xml -L https://raw.githubusercontent.com/raspberry-vanilla/android_local_manifest/android-15.0/manifest_brcm_rpi.xml --create-dirs
3. curl -o .repo/local_manifests/remove_projects.xml -L https://raw.githubusercontent.com/raspberry-vanilla/android_local_manifest/android-15.0/remove_projects.xml
4. repo sync
5. . build/envsetup.sh
6. lunch aosp_rpi4_car-bp1a-userdebug
7. make bootimage systemimage vendorimage -j$(nproc)

### Error Output:
  [ 0% 98/57018] build out/target/product/rpi4/obj/MESON_MESA3D/.build.timestamp
FAILED: out/target/product/rpi4/obj/MESON_MESA3D/.build.timestamp 
/bin/bash -c "(rm -rf out/target/product/rpi4/obj/MESON_MESA3D/ ) && (mkdir -p out/target/product/rpi4/obj/MESON_MESA3D ) && (cd out/target/product/rpi4/obj/MESON_MESA3D && cp -r external/mesa3d/ .) && (mkdir -p out/target/product/rpi4/obj/MESON_MESA3D/build ) && (cd out/target/product/rpi4/obj/MESON_MESA3D && PATH=/usr/local/bin:$PATH meson build --cross-file /home/ubuntu/AOSP_15/out/target/product/rpi4/obj/MESON_MESA3D/cross_file.txt -Dplatforms=android -Dplatform-sdk-version=35 -Dgallium-drivers=vc4,v3d -Dvulkan-drivers=broadcom -Dgbm=enabled -Dgallium-nine=false -Ddri3=enabled -Dglx=disabled -Degl=enabled -Dllvm=disabled -Dgbm-backends-path=/vendor/lib64 -Dlibunwind=disabled -Dcpp_rtti=false -Dandroid-libbacktrace=disabled ) && (ninja -C out/target/product/rpi4/obj/MESON_MESA3D/build ) && (touch out/target/product/rpi4/obj/MESON_MESA3D/.build.timestamp )"
meson is not allowed to be used. See https://android.googlesource.com/platform/build/+/main/changes.md#PATH_Tools for more information.
15:02:32 ninja failed with: exit status 1

#### failed to build some targets (02:45 (mm:ss)) ####

So can you please advice a proper workaround to build this with `meson` Whether Raspberry Pi 4 support via `raspberry-vanilla` needs updates to align with Android 15 build restrictions Any known patches or changes to bypass or correctly integrate `meson` under AOSP 15

KonstaT, 2025-04-11:
Please follow the instructions on how to setup build environment. It also includes steps on installing meson.
https://github.com/raspberry-vanilla/android_local_manifest/tree/android-15.0

bhaskarreddyv123, 2025-04-11:
i followed the same process what you mentioned. But i tried at ubuntu OS 18 version is it compulsory  it would 22 version

KonstaT, 2025-04-11:
See links I added to the previous post.

Ubuntu 22.04 is the recommended build environment. Sure, you can use other distributions to build as well but in that case you might need to figure out yourself how to get the required packages/tools installed.

closed 2025-04-11 by KonstaT