raspberry-vanilla/android_local_manifest #198: unable to boot builded image in rpi5

msarvi, 2025-04-15:
after build AOSP 15, i deploy image to sdcard  of rpi5, but booting proccess  never start just only a black screen comes and restart booting, this is my config.txt-->
`# Kernel
kernel=Image

# Ramdisk
initramfs ramdisk.img followkernel

# Camera
camera_auto_detect=1

# Display
disable_overscan=1

# Display panel
#dtoverlay=vc4-kms-dsi-7inch
#dtoverlay=vc4-kms-dsi-ili9881-7inch

# Graphics acceleration
dtoverlay=vc4-kms-v3d
dtoverlay=cma,cma-320

# I2C
#dtparam=i2c_arm=on

# I2S
#dtparam=i2s=on

# SPI
#dtparam=spi=on

# USB
dtoverlay=dwc2,dr_mode=peripheral
[cm5]
dtoverlay=dwc2,dr_mode=otg
[all]`

and cmdline.txt is ->
`console=ttyAMA10,115200 no_console_suspend root=/dev/ram0 rootwait androidboot.hardware=rpi5 androidboot.selinux=permissive androidboot.hypervisor.vm.supported=1 androidboot.wificountrycode=00`

KonstaT, 2025-04-15:
Yes, I know how `config.txt` and `cmdline.txt` look like in Raspberry Vanilla.

Have you read the project wiki pages? You're expected to use HDMI display that supports 1920x1080 resolution (and reports resolutions it supports via EDID) and boot from a sdcard. Otherwise you need to do something that's described in the wiki.
https://github.com/raspberry-vanilla/android_local_manifest/wiki

There's an open issue on how to debug failing boot process. There's no point in guessing without seeing the appropriate logs.
https://github.com/raspberry-vanilla/android_local_manifest/issues/181

msarvi, 2025-04-15:
@KonstaT , Thanks, after i change setting as in wiki mentions, the boot proccess finished successfully but my screen is blinking (turn screen black and turn on) why?

KonstaT, 2025-04-15:
Most likely a power supply issue. You need to use 5V 5A PSU on Pi 5 or it doesn't power the USB ports properly.

But again, what's the point of me guessing if you don't provide the relevant information i.e. even what type of display are you using in this case.

closed 2025-04-15 by KonstaT