raspberry-vanilla/android_local_manifest #199: [AOSP14] [RPI5] IMU BNO055 integration

Nickldio2, 2025-04-16:
Hi,

I am desperately trying to integrate the 9-axis IMU BNO055.

After loads of unsuccessful tries on my own, I have found your forum post from 2021 [here](https://forums.raspberrypi.com/viewtopic.php?t=297349&start=125#p1927487).

So far I have followed the described instructions, added the BNO055 to the android-i2c-sensor overlay and added it to the config.txt in /boot.

After compiling I can see raw sensor values on /dev/iio:device0, but the android sensorservice still reports 0 sensors. I think I have to properly integrate it to the HAL aswell. In your forum post you mention that your builds use the [iio sensors HAL](https://github.com/[lineage-rpi/android_hardware_intel_sensors-iio](https://github.com/lineage-rpi/android_hardware_intel_sensors-iio)), I guess I have to manually integrate it?

Sorry if there is information im just too stupid to understand, I am quite new to this and honestly don't know where to go from here, does your AOSP source even include the iio sensors HAL or just your release Lineage/AOSP builds?
I assume I need to somehow include the service for the iio sensor HAL to get it to work?

Thanks in advance :)

KonstaT, 2025-04-16:
Yes, you would need to add an Android sensor HAL to use IIO sensors. Goal of the Raspberry Vanilla project is to support the built-in hardware on Raspberry Pi. It doesn't have the drivers/HALs to support every possible peripheral equipment.

Generally speaking https://github.com/lineage-rpi/android_hardware_intel_sensors-iio would be one option.
STMicroelectronics provides much more modern AIDL sensor HAL for their hardware. Last I've tested it also requires to use the specific kernel IIO drivers they also provide (https://github.com/STMicroelectronics/st-mems-android-linux-drivers-iio).

Nickldio2, 2025-04-16:
Okay thanks, sadly I could not find any Bosch specific HAL integration for android except for some automotive sensors, so I will try adding the lineage iio HAL to my build first, will report how it goes 👍

Nickldio2, 2025-04-16:
I have cloned the intel sensors-iio and added the HAL to my VINTF manifest:

```
    <hal format="legacy">
        <name>sensors</name>
        <transport>hwbinder</transport>
        <version>1.0</version>
        <interface>
            <name>ISensors</name>
            <instance>default</instance>
        </interface>
    </hal>
```

but that always gets turned into this on build:

```
    <hal format="hidl">
        <name>sensors</name>
        <transport>hwbinder</transport>
        <fqname>@1.0::ISensors/default</fqname>
    </hal>
```

I guess that's on purpose?

I also added it in device/brcm/rpi5/device.mk:
`PRODUCT_PACKAGES += sensors.iio`

I checked that the package is indeed being built.

I guess the problem is that the HAL is not being loaded and I get "hwservicemanager: Cannot find entry android.hardware.sensors" errors, aswell as "Cannot get virtualdevice_native service" warnings, although I'm unsure if that's related at all.

What am I missing?

Also I am a bit lost what to do with the STMicroelectronics solution, as I understand it I would have rewrite significant portions of the HAL, or was this just meant as an example for how a modern AIDL HAL integration looks like?
In the past I have also tried creating my own integration with AIDL but ended up with the scenario of the sensor being detected by the sensorservice and also displayed in the cpu-z app, but could not get it to deliver data. In the end the FMQ just filled up to 256 entries and the data was somehow not able to be processed by the sensorservice, and I never found out why, so I got frustrated and tried this solution.

KonstaT, 2025-04-17:
You would need to build sensors@1.0 HIDL among other things. See attached patch for reference.
[0001-add-sensors-hal.patch.txt](https://github.com/user-attachments/files/19798174/0001-add-sensors-hal.patch.txt)

> I guess that's on purpose?

Yes.

> Also I am a bit lost what to do with the STMicroelectronics solution

I was providing general references how to implement IIO sensor HAL. As mentioned, sensors@1.0 HIDL has been deprecated and all HALs should be AIDL for Android 15.

Nickldio2, 2025-04-21:
Okay, thanks to your instructions I got it to work, you are awesome!
Very cool that you go out of your way and help people like me :)

Any way to buy you a coffee? I remember you used to have a donation link set up, but I can't find it.

KonstaT, 2025-04-21:
Well done! Good to know you got it working.

We have interesting fund-raising laws where I reside so I can't legally ask for gratuitous donations. I can still accept donations, though (and you did also receive something in return).

You can donate to konsta09@gmail.com on PayPal or send bitcoin to 1KonstafKxLejN74G2NLZhD2TGkA6m1MCD. Thank you.