raspberry-vanilla/android_local_manifest #200: How to enable Network services like Ethernet and WiFi without connecting the display to RPi4 device ?

SagarMorse, 2025-04-23:
Hi Team,

Services like Ethernet and WiFi services are not started without connecting the HDMI display to the RPi4 device with recent additions(After Feb 2025) to android-14.0.0_r67 branch. It looks like there are some changes in HDMI HAL implementation in the last two months. RPi4 android 14 device was fetching IP address to ethernet interface without connecting any display until Jan 2025 but not now anymore. This is required as I was accessing the RPi4 through adb over Ethernet IP. Is there any solution that makes android starts atleast networking services(ethernet and wifi) even if the HDMI display is not connected ?

Thanks in advance.

KonstaT, 2025-04-23:
I'm not a team, it's just me.

I'm not aware of any changes in past couple of months that would have any effect on this. Impossible to say without seeing the appropriate logs (`logcat` in this case).

`android-14.0` branch is pretty much unmaintained and I've only brought in some backports. You should use the latest release (`android-15.0`) at this point.

SagarMorse, 2025-04-23:
Thanks for the reply @KonstaT,  Following is the logcat output. I see that Ethernet service(Log: EthernetServiceImpl: Starting Ethernet service) is started after connecting the display. Attached the logcat output file.

KonstaT, 2025-04-23:
There is no `logcat` attached.

SagarMorse, 2025-04-23:
[resolution.txt](https://github.com/user-attachments/files/19865937/resolution.txt) 

Sorry, it was missing.

closed 2025-04-23 by KonstaT