raspberry-vanilla/android_local_manifest #201: [MultizoneAudio] PoC using 2 HDMI Ports (Primary & Secondary) as 2 zones

hkrishna-git, 2025-04-24:
Hi @KonstaT ,

Good day.
Hope you are doing good.

I would like to contribute to the wiki page for updating the PoC done "MultizoneAudio" on Raspberry Pi 4B.
I have made changes to the markdown file in the wiki project but am unable to push to the remote as write access is disabled.

If possible, could you please spare some time and update the wiki page with the details attached. 

[MultizneAudio-PoC.md](https://github.com/user-attachments/files/19885475/MultizneAudio-PoC.md)

[MultizoneAudioPatches.zip](https://github.com/user-attachments/files/19885492/MultizoneAudioPatches.zip)

I have added the details, please have a review and update the page accordingly.

Thank you for the support.

KonstaT, 2025-04-29:
Thank you for your contribution.

As a proof of concept, I'm not sure what would be the correct place for this. Raspberry Vanilla `android-14.0` branch is in practice unmaintained and discontinued at this point. All the project's wiki pages have been updated to reflect current `android-15.0` behaviour. Your patches also modify AOSP source outside Raspberry Pi specific device configuration that Raspberry Vanilla provides.

I can leave this issue open so it's easier for people interested in multi-zone audio on Automotive target to find.

KonstaT, 2026-11-02:
Closing as outdated as Raspberry Vanilla `android-14.0` branch has been unmaintained for ages and audio HAL implementation has changed since.

closed 2026-11-02 by KonstaT