raspberry-vanilla/android_local_manifest #203: Android 14 for rpi4 Stuck on Boot loop

siddheshp27, 2025-04-28:
Hi @KonstaT ,

I am see this error in the logcat:
`01-01 00:00:24.847   367   367 F libc    : Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 367 (android.hardwar), pid 367 (android.hardwar)
01-01 00:00:24.890     0     0 I servicemanager: Found android.hardware.power.IPower/default in device VINTF manifest.
01-01 00:00:24.900     0     0 I servicemanager: Found android.hardware.cas.IMediaCasService/default in device VINTF manifest.
01-01 00:00:24.907     0     0 W healthd : No battery devices found
01-01 00:00:24.920     0     0 W healthd : battery none chg=
01-01 00:00:24.959     0     0 I         : file system registered
01-01 00:00:24.974     0     0 W         : using random self ethernet address
01-01 00:00:24.978     0     0 W         : using random host ethernet address
01-01 00:00:25.914   571   571 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
01-01 00:00:25.914   571   571 F DEBUG   : Build fingerprint: 'Raspberry/aosp_rpi4/rpi4:14/AP2A.240905.003/eng.ubuntu.20250424.222243:userdebug/test-keys'
01-01 00:00:25.914   571   571 F DEBUG   : Revision: '0'
01-01 00:00:25.914   571   571 F DEBUG   : ABI: 'arm64'
01-01 00:00:25.914   571   571 F DEBUG   : Timestamp: 1970-01-01 00:00:25.261144596+0000
01-01 00:00:25.914   571   571 F DEBUG   : Process uptime: 3s
01-01 00:00:25.914   571   571 F DEBUG   : Cmdline: /vendor/bin/hw/android.hardware.usb.gadget-service.rpi
01-01 00:00:25.914   571   571 F DEBUG   : pid: 367, tid: 367, name: android.hardwar  >>> /vendor/bin/hw/android.hardware.usb.gadget-service.rpi <<<
01-01 00:00:25.914   571   571 F DEBUG   : uid: 1000
01-01 00:00:25.914   571   571 F DEBUG   : tagged_addr_ctrl: 0000000000000001 (PR_TAGGED_ADDR_ENABLE)
01-01 00:00:25.914   571   571 F DEBUG   : signal 6 (SIGABRT), code -1 (SI_QUEUE), fault addr --------
01-01 00:00:25.914   571   571 F DEBUG   :     x0  0000000000000000  x1  000000000000016f  x2  0000000000000006  x3  0000007fe7092530
01-01 00:00:25.915   571   571 F DEBUG   :     x4  0000000000000010  x5  0000000000000010  x6  0000000000000010  x7  7f7f7f7f7f7f7f7f
01-01 00:00:25.915   571   571 F DEBUG   :     x8  00000000000000f0  x9  00000076606c7390  x10 ffffff80ffffffdf  x11 fffffffffffffffd
01-01 00:00:25.915   571   571 F DEBUG   :     x12 0000007fe7092050  x13 000000000000001c  x14 0000007fe7092188  x15 0000000052c63089
01-01 00:00:25.915   571   571 F DEBUG   :     x16 0000007660786fe0  x17 000000766076a860  x18 0000007663f68000  x19 000000000000016f
01-01 00:00:25.915   571   571 F DEBUG   :     x20 000000000000016f  x21 00000000ffffffff  x22 0000007fe7092718  x23 0000000000000000
01-01 00:00:25.915   571   571 F DEBUG   :     x24 0000000000000000  x25 0000000000000000  x26 0000000000000000  x27 0000000000000000
01-01 00:00:25.915   571   571 F DEBUG   :     x28 0000000000000000  x29 0000007fe70925b0
01-01 00:00:25.915   571   571 F DEBUG   :     lr  000000766070aad8  sp  0000007fe7092510  pc  000000766070ab08  pst 0000000000000000
01-01 00:00:25.915   571   571 F DEBUG   : 4 total frames
01-01 00:00:25.915   571   571 F DEBUG   : backtrace:
01-01 00:00:25.915   571   571 F DEBUG   :       #00 pc 0000000000066b08  /apex/com.android.runtime/lib64/bionic/libc.so (abort+168) (BuildId: 995c9ec5a72bec30d188a89027728cd8)
01-01 00:00:25.915   571   571 F DEBUG   :       #01 pc 0000000000011300  /vendor/bin/hw/android.hardware.usb.gadget-service.rpi (aidl::android::hardware::usb::gadget::UsbGadget::UsbGadget()+100) (BuildId: 656eec0012c7f13cc97c5f1eab82d2d3)
01-01 00:00:25.915   571   571 F DEBUG   :       #02 pc 0000000000010ab4  /vendor/bin/hw/android.hardware.usb.gadget-service.rpi (main+56) (BuildId: 656eec0012c7f13cc97c5f1eab82d2d3)
01-01 00:00:25.915   571   571 F DEBUG   :       #03 pc 0000000000060748  /apex/com.android.runtime/lib64/bionic/libc.so (__libc_init+116) (BuildId: 995c9ec5a72bec30d188a89027728cd8)
01-01 00:00:25.928     0     0 I init    : Service 'vendor.usb_gadget-default' (pid 367) received signal 6
01-01 00:00:25.936     0     0 I init    : Sending signal 9 to service 'vendor.usb_gadget-default' (pid 367) process group...
01-01 00:00:25.946     0     0 I libprocessgroup: Removed cgroup /sys/fs/cgroup/uid_1000/pid_367`

I haven't made any changes, I even tried using different releases for android 14 but I was able to see similar problems 

Here is the complete log file: 

[logcat.txt](https://github.com/user-attachments/files/19939144/logcat.txt)

KonstaT, 2025-04-29:
Duplicate of https://github.com/raspberry-vanilla/android_local_manifest/issues/200.

`android.hardware.usb.gadget-service.rpi` crashing once has nothing to do with this as explained in https://github.com/raspberry-vanilla/android_local_manifest/issues/174#issuecomment-2598093167. And this also no longer happens on `android-15.0` as the service start later after switching to APEX.

This is related to HDMI-CEC and/or your display setup. I can't see in the complete logcat that hwcomposer detects any display. What is your display setup?

As mentioned in https://github.com/raspberry-vanilla/android_local_manifest/issues/200 I can't reproduce this with/without HDMI display connected. Also `android-14.0` branch is pretty much unmaintained at this point and you should be using `android-15.0` by now.

siddheshp27, 2025-04-29:
I am using HDMI cable to connect to display, I am able to see android boot logo on the display but its stuck on that logo, and I had tried running android 14 previously( in the month of december) but I did not faced any issues then, should I recapture the logs and add them?

KonstaT, 2025-04-29:
Did you read https://github.com/raspberry-vanilla/android_local_manifest/issues/200 or the above reply? This is related to HDMI-CEC and/or your display setup.

As also mentioned above, you can consider `android-14.0` branch unmaintained and discontinued at this point. Also, I can't reproduce this on `android-14.0`. So, rather update to using `android-15.0` and attach the logs if you can reproduce this there.

KonstaT, 2025-05-08:
Closing as `android-14.0` branch is no longer maintained. I'm not able to reproduce this issue on `android-14.0`.

closed 2025-05-08 by KonstaT