raspberry-vanilla/android_local_manifest #205: Bluetooth cycle on rpi5 with AOSP 15 in car mode

yungblood, 2025-06-06:
I followed the directions for building a car image for rpi5. Everything went relatively smooth.
However, when I link to my Samsung Galaxy Z Fold 5, the rpi gets stuck in loop
connecting...
turns off bluetooth...
turns bluetooth back on...
connecting...

KonstaT, 2025-06-07:
Please attach the appropriate logs (`logcat` in this case).

I can't reproduce this the last I've tested. Try toggling the settings option on your phones bluetooth settings.
https://github.com/raspberry-vanilla/android_local_manifest/issues/110#issuecomment-2733524508

Also see the previous report of similar issues.
https://github.com/raspberry-vanilla/android_local_manifest/issues/33

yungblood, 2025-06-09:
Ok, I was able to turn off sharing messages on my phone, and it stopped cycling.
However, my phone doesn't acknowledge it's in Android Auto mode. 
I am able to play music and have it come out the display on my rpi5, but it shows it as bluetooth audio instead of the actual android auto app I used. (in this case Amazon Music)

closed 2025-06-12 by KonstaT