raspberry-vanilla/android_local_manifest #210: Video freezes on loop when using FFmpeg decoder

cwall-dev, 2025-07-11:
I created a simple website that plays a video in an infinite loop using JavaScript.

I'm testing the website using the built-in WebView Shell app.

With enabled FFmpeg decoder (`persist.vendor.ffmpeg_codec2.rank.audio=16` and `persist.vendor.ffmpeg_codec2.rank.video=256`), the video usually freezes at the end of the first loop. As soon as I disable FFmpeg (`persist.vendor.ffmpeg_codec2.rank.audio=0xFFFFFFFF` and `persist.vendor.ffmpeg_codec2.rank.video=0xFFFFFFFF`), the loop works flawlessly.

When the freeze occurs, logcat shows the following error messages:
``` text
07-11 09:16:16.279   941  1041 I MediaFocusControl: abandonAudioFocus() from uid/pid 10057/1999 clientId=android.media.AudioManager@ac02c11org.chromium.content.browser.AudioFocusDelegate@7662176 callingPack=org.chromium.webview_shell
07-11 09:16:16.281  1999  2126 D CCodecBufferChannel: [c2.ffmpeg.h264.decoder#152] MediaCodec discarded an unknown buffer
07-11 09:16:16.281  1999  2126 D CCodecBufferChannel: [c2.ffmpeg.h264.decoder#152] MediaCodec discarded an unknown buffer
07-11 09:16:16.281   941  1041 I MediaFocusControl: requestAudioFocus() from uid/pid 10057/1999 AA=USAGE_MEDIA/CONTENT_TYPE_UNKNOWN clientId=android.media.AudioManager@ac02c11org.chromium.content.browser.AudioFocusDelegate@7662176 callingPack=org.chromium.webview_shell req=1 flags=0x0 sdk=29
07-11 09:16:16.281   941  1041 I AS.FadeOutManager: unfadeOutUid() uid:10057
07-11 09:16:16.281  1999  2063 D MediaCodec: keep callback message for reclaim
07-11 09:16:16.282  1999  2126 I CCodecConfig: query failed after returning 7 values (BAD_INDEX)
07-11 09:16:16.295   941   995 I DisplayDeviceRepository: Display device changed render timings: "Built-in Screen", renderFrameRate=60.0, presentationDeadlineNanos=16666667, appVsyncOffsetNanos=1000000, frameRateOverrides=[]
07-11 09:16:16.296   776  2133 D C2FFMPEGVideoDecodeComponent: onFlush_sm
07-11 09:16:16.301  1999  2140 D CCodecBufferChannel: [c2.ffmpeg.h264.decoder#152] Ignoring stale input buffer done callback: last flush index = 1803, frameIndex = 1803
07-11 09:16:16.353  1999  2069 E m.webview_shell: Transaction::setFrameRate failed - invalid frame rate inf
```

Disabling FFmpeg seems to resolve the issue entirely.
Any idea what might be causing this behavior?

KonstaT, 2025-07-11:
Which branch? `rpi5` I assume?

I can't reproduce with local H.264 files and built-in media player at least.

KonstaT, 2025-07-11:
Which branch?

On `rpi5` FFMpeg is only there to support H.265 hardware decoding so you might as well disable it.

cwall-dev, 2025-07-11:
I'm currently testing with the android-16.0 branch.
These are my current FFmpeg-related properties:
```
persist.vendor.ffmpeg_codec2.rank.audio=16
persist.vendor.ffmpeg_codec2.rank.video=256
persist.vendor.ffmpeg_codec2.v4l2.h265=true
```

Is there a way to configure the system so that FFmpeg is only used for H.265 hardware decoding, and all other formats (like H.264) fall back to the default Android decoder?

cwall-dev, 2025-07-11:
Okay, I will give it a try and then check whether the ffmpeg decoder behaves differently with H265 content.